keptenkurk/BS440 #13: How to use auth_google.py on Rpi without browser

keptenkurk, 2016-08-03:
In order to generate the token auth_google.py needs to be run by
python auth_google.py [GOOGLE CLIENT ID] [GOOGLE CLIENT SECRET] https://www.googleapis.com/auth/fitness.body.write

is it possible to perform this step on a Rpi without browser?

managementboy, 2016-08-04:
No, I used vncserver just one time to do this.

keptenkurk, 2016-08-04:
Got it! Finaly googled my way through it. Here's the recipe:

Phase 1. Create a project in [Google Developers Console](https://console.developers.google.com/flows/enableapi?apiid=fitness) and enable the fitness API. 
Create an OAuth client ID with "Other UI" type and select "User data". Note the client_id and client_secret.

Phase 2. Using your editor create a URL like:
https://accounts.google.com/o/oauth2/v2/auth?scope=https://www.googleapis.com/auth/fitness.body.write&redirect_uri=urn:ietf:wg:oauth:2.0:oob&response_type=code&client_id={client-id}&access_type=offline
where you replace {client_id} with your own from Phase 1.
(A client_id looks like: 123456789012-rk5cgggoup1253gd72jd82js82js82s.apps.googleusercontent.com)
Paste the full URL in the browser and hit enter.
The trick here is that the redirect_uri is defined as "urn:ietf:wg:oauth:2.0:oob" which tells Google to present the Athentication code in an easy copy/paste window. 

Phase 3. It is not the Authentication code we need, it's the json with access_token and refresh_token (especially the refresh_token which does not age)
To get these we need to POST our data to yet another API. 
Use POSTMAN from the Google webstore (or curl) to post to this URL:
  URL: https://www.googleapis.com/oauth2/v3/token?
with these these key/values:
  code: {Authentication code from Phase 2}
  redirect: urn:ietf:wg:oauth:2.0:oob
  client_id: {client_id from Phase 1} 
  client_secret: {client_secret from Phase 1}
  scope: https://www.googleapis.com/auth/fitness.body.write
  grant_type: authorization_code

When using curl this needs to be crafted into a single string like:
https://www.googleapis.com/oauth2/v3/token?code=4/utnSXJGuABllCl8j4NDyI_5-ItEy8v84f6yptpMJ0co&redirect_uri=urn:ietf:wg:oauth:2.0:oob&client_id=36273628994738-rk5cgggoup128hs7g9sl927h1vjc349q.apps.googleusercontent.com&client_secret=dhTfda67khU789lhGfg986yw&scope=https://www.googleapis.com/auth/fitness.body.write&grant_type=authorization_code

Google will respond with the json file which then can be saved a copied to your Pi.

(ps: don't bother trying this data, it's all fake)

keptenkurk, 2016-08-04:
Hmmm... my json appears to have an invalid structure as it halts:
Traceback (most recent call last):
  File "BS440.py", line 171, in <module>
    UpdateGoogle(config, persondata, weightdatasorted)
  File "/home/pi/BS440google.py", line 112, in UpdateGoogle
    googleClient = GetGoogleClient(googleauthfile)
  File "/home/pi/BS440google.py", line 28, in GetGoogleClient
    credentials = Storage(filename).get()
  File "/usr/local/lib/python2.7/dist-packages/oauth2client/client.py", line 409, in get
    return self.locked_get()
  File "/usr/local/lib/python2.7/dist-packages/oauth2client/file.py", line 65, in locked_get
    credentials = client.Credentials.new_from_json(content)
  File "/usr/local/lib/python2.7/dist-packages/oauth2client/client.py", line 304, in new_from_json
    module_name = data['_module']
KeyError: '_module'

This is what my authentication json looks like:
{
  "access_token": "----------some very long string-----------------",
  "token_type": "Bearer",
  "expires_in": 3600,
  "refresh_token": "-----------another very long string--------------"
}

Would that be the correct structure?

managementboy, 2016-08-09:
looks the same in mine. I will create a new pull request, as I have refactored a bit more code.

keptenkurk, 2016-08-10:
That's odd... the KeyError message seems to hint that the json should contain some '_module' key in it?
Pretty lost here.

managementboy, 2016-08-11:
sorry to hear. I have been updateing my weight twice a day for a week now without fail. Hope you find out what the reason is. Maybe some missing dependency? The Fitbit code I used as inspiration had these: 
**oauthlib == 1.0.3**
**google_api_python_client == 1.5.0**
cherrypy == 5.4.0
python_dateutil == 2.5.3
httplib2 == 0.9.2
**requests_oauthlib == 0.6.1**
PyYAML == 3.11

keptenkurk, 2016-08-13:
Finally was able to generate a working google.json.
After a fruitless attempt to get the desktop running on the Pi, i choose the other way around and ran the auth_google.py on a windows platform instead (this requires to run pip install --upgrade oauth2client). auth_google.py created a different json which indeed contains the information above but also the required _module parameter.
This eventually lead to the error message: "invalid_clientUnauthorized" which was resolved by [this](http://stackoverflow.com/questions/17166848/invalid-client-in-google-oauth2). Or in short: The project name should be set the same as the product name in the Consent screen section of the Google Developers Console.

So to me this is a usable work around to the question posted.

closed 2016-08-13 by keptenkurk