keptenkurk/BS440 #17: Logrotation

keptenkurk, 2016-09-20:
Rotate logfiles as the tend to become pretty big over time. 
(from http://www.domoticz.com/forum/viewtopic.php?f=38&t=11305)

remb0, 2016-09-23:
also
/var/daemon.log
because of the bleutooth logging.

keptenkurk, 2016-09-23:
Well... than have a look at syslog, debug and auth.log too... they are also cluttered because of this bluetooth restarting every 10 seconds. The good thing is: all these files will be rotating and zipped already by the OS. Still it would be nicer if it were possible to keep the bluetooth running. BTW i schedule this pi to be only active an hour or so in the morning :-)

keptenkurk, 2017-02-28:
With the solution of #32 and no BT adapter resets the /var/log/daemon.log is not cluttered anymore. Also by moving the log outputs in py
https://guides.github.com/features/mastering-markdown/gatt's gatttool.py scan procedure from INFO to DEBUG there will be no unwanted logging. Therefore logrotation won't be needed.

closed 2017-02-28 by keptenkurk

DjZU, 2023-04-11:
Since then I did 2 things to prevent the log to become big.

1. Run BS440 without privileges to prevent adapter reset, as described here: https://github.com/keptenkurk/BS440/issues/110
2. Set pygatt log level to WARNING by amending BS440.py code as follows:
```
	plugins = {}

+++	# Set pygatt log level
+++	# INFO floods the log file with 'Starting BLE scan', 'Discovered <device>' and 'Found <devices_count> BLE devices'
+++	# Possible values: NOTSET, DEBUG, INFO, WARNING, ERROR, CRITICAL
+++	logging.getLogger('pygatt').setLevel(logging.WARNING)

	# set up logging
```