keptenkurk/BS440 #20: Settings in .ini

remb0, 2016-09-21:
IN BS440domoticz.py
unit Hardwareid and unit are hardcoded. can this be placed in the ini?

the same for google auth disable/enable.
when not using google, bs440 will import bs440google always, even when not using.

keptenkurk, 2016-09-21:
Could you enlighten us on the purpose of these ID's. No porblem to get these from the .ini
Do we need to take care of not creating a virtual sensor with a conflicting ID? Limited Domoticz knowledge causes me being in the dark here.

remb0, 2016-09-21:
in   domoticz.py you use:
      url = 'http://%s/json.htm?type=command&param=udevice&hid=**13**&' \
              'did=%s&dunit=**4**&dtype=93&dsubtype=1&nvalue=0&svalue=%s' % (

but the hid and unitid can for every user different.    and you have a great ini file to be modulair, so it would better to use the ini for this parameters to be more flexible.

keptenkurk, 2016-09-29:
Added feature in commit c140a5c3375d3da90b799b76358cf3ed59471840

closed 2016-09-29 by keptenkurk