keptenkurk/BS440 #21: domoticz isn't updating

remb0, 2016-09-21:
log is showing:
UpdateDomoticz Updating Domoticz for user Remco at index 1106 with weight 72.7

but domoticz device isn't updating (mail works).

So I checked ini settings.  in the ini i used the internal ip with port   192.168.0.150:8080
also 127.0.0.1:8080
I use form authentication and excluded username and password for internal uses.

I tested also with basic authentication in domoticz, same result

my devices:
1106	Virtual	61	4	Weegschaal Remco	Weight	BWR102	75.0 kg
1105	Virtual	60	4	Weegschaal Annika	Weight	BWR102	75.0 kg

my config in ini:
[Person2]
username: Remco
useremail: xxxx@gmail.com
domoticz_idx: 1106
#googleauthfile: google-remco.json

in bs440domoticz.py

        url = 'http://%s/json.htm?type=command&param=udevice&hid=13&' \
              'did=%s&dunit=4&dtype=93&dsubtype=1&nvalue=0&svalue=%s' % (
hid 13 correspond with my virtual hardware id

keptenkurk, 2016-09-21:
It looks like you're using the Domoticz Idx (first column in the device list on Domoticz).
However, you should be using the ID instead (the third column), the 60 and 61 value in your case.

You could verify by calling the url from your browser and check for OK result.
When run on the same network from your browser,
http://192.168.0.150:8080/json.htm?type=command&param=udevice&hid=13&did=61&dunit=4&dtype=93&dsubtype=1&nvalue=0&svalue=75.5
(all on a single line ofc)

If this works it's clearly my fault by designating it "domoticz_idx" in the .ini instead of "domoticz_ID"

remb0, 2016-09-21:
nice!!!   my fault,   stupid.      
but Idx stands for the first column in domoticz,  so i thought I was right.
maybe you can update your great tutorial to be more specific? or an extra comment in the ini file.  (for others)

keptenkurk, 2016-09-29:
Renamed IDX to ID to avoid future confusion. f395fb5d2a849ccb157343235da497e4722e39fb

closed 2016-09-29 by keptenkurk