keptenkurk/BS440 #24: BS440 as service?

remb0, 2016-10-02:
when this fantastic script is running as an service, monit can monitoring it, 
plus it's restartable etc.

I have tried to modify: 
https://www.domoticz.com/wiki/Presence_detection_(Bluetooth_4.0_Low_energy_Beacon)
But don't get it work?
How do you run it automaticly after boot?

keptenkurk, 2016-10-04:
The nicest way is definately to run it as a service. I choose the rude way by starting the script through cron at boot. Just put next line as first line in your cron (edit with `sudo crontab -e`)
`@reboot sh /home/pi/BS440.sh > /home/pi/cronlog 2&1`

DjZU, 2017-02-28:
I run it at boot through /etc/rc.local with 
`su -l <user> -c "/path/to/BS440/BS440.sh >/dev/null 2>&1 &"`
But if you did not set capabilities to hcitool (see my comment in #32 ) you'll need to be a super user:
`/path/to/BS440/BS440.sh >/dev/null 2>&1 &`
My system keep running several days without any reboot and the script does not hang then so far I'm happy with that way of firing the script.
If you really want it to be a service, you should have a look [there](url).

closed 2017-03-18 by keptenkurk