keptenkurk/BS440 #28: Questions / Improvements ?!

Tristan79, 2016-12-23:
# Questions

 * Can you also set the heigh, active(?) & age?
 * What about the person?
 * It would be nice to set your age to +1 on your birthday
 * And does this frequent scanning affect the battery life of other small devices?
 * It should be possible to set the date, please implement as a def ;-)

keptenkurk, 2016-12-23:
@Tristan79 Do you (or someone else) know if this is at all possible with the Medisane app? In that case it could be made but i've never seen it. If not: 
* setting of height, active and age can only be set at the scale.
* the scale chooses the person by comparing the weight form last weighings. If in doubt it will show the persons to select from
* That would require setting it by bluetooth (see first request)
* If in the vincinity of the scale that could well be. For me: the scanning is only active at the times i'm supposed to weigh myself in the morning
* unsure what you mean by setting the date. The system date is taken and qt every request the scale's date will sync to that. Is that what you mean?

Tristan79, 2016-12-23:
First if all, I didn't use or install the app at all to setup (and the manual online was very slim)! 
Actually I bought it because I saw you page after a fiasco with withers scale. I am for no-cloud smart home!  So I am very happy this works! :-) 

> If in the vincinity of the scale that could well be. For me: the scanning is only active at the times i'm supposed to weigh myself in the morning.

Well going to test that out by running it for a couple of days, I have a [plant sensor](http://domoticz.com/forum/viewtopic.php?f=56&t=13306&hilit=mo+flora)which you can only poll once every hour not to drain the battery, lets see what happens :-) Hope the low energy bluetooth is very efficient in sending out its id every x seconds...  probably after that start it when the occupy roomsensor is triggered... and kill it when its off :-) 

> unsure what you mean by setting the date. The system date is taken and qt every request the scale's date will sync to that. Is that what you mean?

Aha, I was wondering how that worked... how it jump from no time in the read out of weight to timed reads out, I thought it was because I while playing to get it to work, executed many of the gatttools command line scattered around the pages i could find. And i saw it mentioned in the source... Didn't know it worked automatically. 

> setting of height, active and age can only be set at the scale.

I am going to have a look with:
https://itunes.apple.com/us/app/lightblue/id639944780?mt=12
If its really this easy to read more info :-)  Anyway I looked a little bit around, maybe [link](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/plugin.xml#L97) there is a way to read it...  but I am just guessing :-) If its really this simple to read out medisana ble devices i am going to look through there products... maybe the have some cool other stuff to integrate into domoticz! But I have no idea how this BLe works. So if you have any good links :-)

Anyway Thanks!

Tristan79, 2016-12-23:
Seems it is possible for other devices... very excited!

https://www.benjamin-schieder.de/blog/2015/05/18/reading-steps-from-a-medisana-vifit-connect-pedometer.html

keptenkurk, 2016-12-23:
@Tristan79 
> maybe link there is a way to read it...
Reading the configured height, age etc is implemented. Writing is not possible as i know of

> But I have no idea how this BLe works
Neither did i when starting this project :-)

BTW: did you check my pages at https://keptenkurk.wordpress.com/2016/02/07/connecting-the-medisana-bs440-bluetooth-scale/ ?

closed 2017-01-22 by keptenkurk