keptenkurk/BS440 #32: Update to latest Pygatt

keptenkurk, 2017-01-22:
Note to self:
Pygatt got some updates recently adressing hcitool lescan. [This commit](url=https://github.com/peplin/pygatt/commit/465c81953126d9678636aa4cb21040cdadde3f23) could possible avoid the adapter reset in 
```def wait_for_device(devname):
    found = False
    while not found:
        try:
            found = adapter.filtered_scan(devname)
            # wait for scale to wake up and connect to it
        except pygatt.exceptions.BLEError:
            adapter.reset()
    return```
as the filtered_scan in fact runs hcitool lescan in background. Would reduce the number of errors in the log.

remb0, 2017-01-26:
does it help? I tried to update but can't update it...

DjZU, 2017-02-28:
Instead of trying to upgrade, I manually applied Pygatt PR [#92](https://github.com/peplin/pygatt/pull/92/commits/8e783c0a0bbfca463eb5883f09f2de6d8659ca86).

As suggested I ran the following command so now I can run hcitool thus BS440.py without super user privileges.
	`` sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hcitool` ``
	`` getcap `which hcitool`  ``
	
Did the same for btmgmt (and removed 'sudo' in btmgmt call from BS440.py)
	`sudo setcap 'cap_net_raw,cap_net_admin+eip' /usr/bin/btmgmt`
	`getcap /usr/bin/btmgmt`

Make backup
	`sudo cp /usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py /usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py.backup`
	
Open
	`gksudo gedit /usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py`
	
Add commit including 'import signal' 

I also commented out these two lines
> 	#log.info("Starting BLE scan")
> 	#log.info("Found %d BLE devices", len(devices))

Now BS440.log does not get cluttered anymore.

keptenkurk, 2017-02-28:
Cool! And thanks for trying. Assumed that this pygatt lescan fix was already in release 3.0.0 and forgot about it. ERROR's in a log are misleading and make people nervous. Thanks for all your contributions.

closed 2017-02-28 by keptenkurk

DjZU, 2017-02-28:
Assumed the same actually but digging in it I decided to make this 'hack' since I'm running this script on a machine which is running all the time not like you if I understood well so the log was more of a problem in my case. I believe some errors can still come when resetting but I did not commented out log.error lines. Thanks for your very good starting point work and maintenance!