keptenkurk/BS440 #33: scan Unexpected error when scanning

davidwul, 2017-02-10:
I tried the code on a ubuntu (16.10).

As soon as I lauch the python script, the bluetooth adapter goes dead and I can't reinit it exept if I reboot.. When lauch the script here are the log messages:
Fri, 10 Feb 2017 22:35:01 INFO     run Running...
Fri, 10 Feb 2017 22:35:01 ERROR    scan Unexpected error when scanning
Fri, 10 Feb 2017 22:35:01 INFO     scan Starting BLE scan
Fri, 10 Feb 2017 22:35:01 ERROR    scan Unexpected error when scanning
Fri, 10 Feb 2017 22:35:01 INFO     scan Starting BLE scan
Fri, 10 Feb 2017 22:35:01 ERROR    scan Unexpected error when scanning
Fri, 10 Feb 2017 22:35:12 INFO     scan Starting BLE scan
Fri, 10 Feb 2017 22:35:12 ERROR    scan No BLE adapter found
Fri, 10 Feb 2017 22:35:22 INFO     scan Starting BLE scan
Fri, 10 Feb 2017 22:35:22 ERROR    scan No BLE adapter found

Any idea what can I do?

thanks

keptenkurk, 2017-02-11:
Pygatt is no more than a shell around both the gatttool and hcitool commandline utilities (the latter called with "hcitool lescan" to find out if the scale is active). Both utilities come with the installation of the [Bluez](http://www.bluez.org/download/) Bluetooth stack. It would be interesting to know which version you are using and if it is possible to manually connect to the scale.

davidwul, 2017-02-12:
I'm running bulez 5.41, pygatt 3.0
What would be the steps to connect to it mannually?

keptenkurk, 2017-02-12:
You would need to see a BT device responding to  when issued on the commandline
'''
hcitool lescan
'''
If your scale responds you should be able to connect to it with gatttool in iteractive mode (hence -I)  like:
'''
gatttool -i hci0 -b 20:C3:8F:E9:56:F1 -I
'''
...with the proper MAC address of your scale. See also the [blog](https://keptenkurk.wordpress.com/2016/03/26/connecting-the-medisana-bs440-bluetooth-scale-part-2/). I must admit my installation is still on pygatt 2.1.0. Need to find some time to redo the whole thing with up-todate versions of Bluez and Pygatt.

davidwul, 2017-02-14:
I try the above steps. So I can scan and see the scale. But with gatttool, I don't think I can connect. 
Regarding the command to trigger the scale to push data, the first command hang with now answer. Only if I issue the last one, I have only a message back : "Characteristic value was written successfully" but nothing else. 
Do you think its related to the gatttool version?

keptenkurk, 2017-02-15:
I think it should work with that version of gatttool too. Could you try to connect to it with the interactive mode of gatttool? with
```gatttool -i hci0 -b 20:C3:8F:E9:56:F1 -I```
and see if you can connect to it manually and try some of the commands? 
It is the interactive mode gatttool uses to talk to the Bluez stack. If you can't connect, maybe you need to set the BLE mode first as i described here: https://keptenkurk.wordpress.com/2016/03/26/connecting-the-medisana-bs440-bluetooth-scale-part-2/

keptenkurk, 2017-02-26:
@davidwul Did you have any success? I did a complete new install on RPi2 with latest Jessie, Bluez 5.44 and Pygatt 3.0.0
I now recall the soure of the "Unexpected error when scanning" error.
When issuing ```adapter.filtered_scan(devname)``` 
Pygatt will run ```hcitool lescan``` with superuser privileges. After a 10 second timeout and the searched device not found hcitool will be killed but leaves the Bluetooth adapter in a undefined state. A second call to hcitool will fail with "Set scan parameters failed: Input/output error" which Pygatt returns to us as  "Unexpected error when scanning". The ony way to recover from this is resetting the adapter with ```adapter.reset()```. Therefore the complete routine of scanning (or waiting for the scale to shuw up) becomes:
```def wait_for_device(devname):
    found = False
    while not found:
        try:
            found = adapter.filtered_scan(devname)
            # wait for scale to wake up and connect to it
        except pygatt.exceptions.BLEError:
            adapter.reset()
    return```
This will show this sequence in the log:
```
Sat, 25 Feb 2017 23:31:28 INFO     scan Starting BLE scan       --> first hcitool lescan
Sat, 25 Feb 2017 23:31:39 INFO     scan Found 0 BLE devices  -->  no device in 10 sec
Sat, 25 Feb 2017 23:31:39 INFO     scan Starting BLE scan       --> let's try gain
Sat, 25 Feb 2017 23:31:39 ERROR    scan Unexpected error when scanning --> adapter fails, reset it!
```

davidwul, 2017-02-27:
Here is what I have from sudo hcitool lescan:
`CF:5F:ED:58:86:6C 0202B66C8658ED5FCF`
`CF:5F:ED:58:86:6C (unknown)`

The reset of the adapter is already in the code of BS440.py. So even with the `adapter.reset()` once I launch the BS440.py, the the bt adapter goes not found, I need to reboot the computer to solve the problem.

I will try again the interactive gattool and post the result here.

davidwul, 2017-02-27:
well here is the result:
`[CF:5F:ED:58:86:6C][LE]> connect`
`Attempting to connect to CF:5F:ED:58:86:6C`
`Error: connect error: Connection refused (111)`

the connection refuse come only when the scale "shut down".

Any idea what to do?

keptenkurk, 2017-02-27:
You mean that right after you stepped on the scale you can connect succesfully? If not it looks like the adapter is not in LE mode. You need to issue 
```
btmgmt le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
```
you can find btmgmt in the tools folder of Bluez. To use it from BS440.py you can put a copy of it in /usr/local/bin

keptenkurk, 2017-02-28:
Another thing comes to mind: do you have su rights when running BS440.py ?
i.e.  sudo python BS440.y

davidwul, 2017-02-28:
yes I have enable le and the result is close to yours:

```
sudo btmgmt le on
hci0 Set Low Energy complete, settings: powered connectable discoverable bondable ssp br/edr le secure-conn 
```

Even without having the device arround, if I launch BS440.py (with or without sudo) I have the following log:

```
Tue, 28 Feb 2017 10:41:16 DEBUG    start gatttool_cmd=gatttool -i hci0 -I
Tue, 28 Feb 2017 10:41:16 INFO     run Running...
Tue, 28 Feb 2017 10:41:16 INFO     scan Starting BLE scan
Tue, 28 Feb 2017 10:41:16 ERROR    scan Unexpected error when scanning
Tue, 28 Feb 2017 10:41:16 INFO     wait_for_device <class 'pygatt.exceptions.BLEError'>
Tue, 28 Feb 2017 10:41:16 INFO     scan Starting BLE scan
Tue, 28 Feb 2017 10:41:16 ERROR    scan Unexpected error when scanning
Tue, 28 Feb 2017 10:41:16 INFO     wait_for_device <class 'pygatt.exceptions.BLEError'>
Tue, 28 Feb 2017 10:41:27 INFO     scan Starting BLE scan
Tue, 28 Feb 2017 10:41:27 ERROR    scan Unexpected error when scanning
Tue, 28 Feb 2017 10:41:27 INFO     wait_for_device <class 'pygatt.exceptions.BLEError'>
Tue, 28 Feb 2017 10:41:37 INFO     scan Starting BLE scan
Tue, 28 Feb 2017 10:41:37 ERROR    scan No BLE adapter found
Tue, 28 Feb 2017 10:41:37 INFO     wait_for_device <class 'pygatt.exceptions.BLEError'>
Tue, 28 Feb 2017 10:41:48 INFO     scan Starting BLE scan

```
As you can see, after a few second, the adapter goes not found.
I tried this script on 3 differents machines (always with ubuntu 16.10 or 16.04) and always had this result. It is maybe due to some ubuntu specificities?

DjZU, 2017-02-28:
@davidwul I run this code on Intel NUC5i3RYB operating x86_64 Ubuntu 16.04.2 LTS. 
Can you provide output of following command?
`echo $(lsb_release -a 2>/dev/null | grep 'Description' | awk -F'\t' '{print $2 }' && uname -i && uname -r) && echo $(dpkg-query -W -f='${binary:Package}\t${Version}\t${Architecture}' bluez) && pip list --format=legacy | grep pygatt`

For me it ouputs:

> Ubuntu 16.04.2 LTS x86_64 4.4.0-57-lowlatency
bluez 5.37-0ubuntu5 amd64
pygatt (3.0.0)

davidwul, 2017-02-28:
Here is what i have:

```
Ubuntu 16.10 x86_64 4.8.0-39-generic
bluez 5.41-0ubuntu3 amd64
pygatt (3.0.0)
```

i had to remove the --format=legacy. It wasn't supported . (no such option: --format)

DjZU, 2017-02-28:
Mmm. I messed up a bit trying to connect to another scale before getting a BS444 but I kept notes of what I did. Try this one:

Open bluetooth configuration file
	`gksudo gedit /etc/bluetooth/main.conf`

Add the following lines at the end of the file

> 		EnableLE = true           // Enable Low Energy support. Default is false.
> 		AttributeServer = true    // Enable the GATT attribute server. Default is false.

Restart bluetooth service
	`sudo service bluetooth restart`

DjZU, 2017-02-28:
I forgot to mention that I'm actually running Ubuntu Studio (that explains the LTS and the lowlatency kernel). 
A major difference is the desktop manager: I have XFCE while you probably have Unity.
XFCE uses blueman as applet while Unity uses default GNOME bluetooth indicator. 
Have a look [there](https://askubuntu.com/questions/777199/bluetooth-le-and-ubuntu-16-04-sometimes-works-other-times-not).

davidwul, 2017-02-28:
Thanks a lot! Editing /etc/bluetooth/main.conf as you sugested made the trick.
This is solve for me. Maybe these instructions should go in the README.

closed 2017-02-28 by davidwul

DjZU, 2017-02-28:
@davidwul Glad to help!

keptenkurk, 2017-03-01:
> Maybe these instructions should go in the README
@davidwul could you drop me a few lines regarding the Ubuntu specifics? I will add that to the Readme.