keptenkurk/BS440 #40: Support more variations of devices, e.g. BS444

lenucksi, 2017-02-17:
Hey,
I wonder if there is any knowledge or attempt to provide support for more scales, such as e.g. the BS444 (http://www.medisana.de/exklusive-angebote/BS-444-ViFit-MX3-Koerperanalysewaage-Aktivitaets-Tracker-B-Ware2.html) or the Sanitas SBF 70 devices (http://www.sanitas-online.de/web/en/products/weight/scales/SBF-70-Bluetooth%C2%AE---Glas-Diagnosewaage)

I've tried to record and make sense of the data sent by a SBF 70 device, however was not too successful in making sense of it...

keptenkurk, 2017-02-17:
@lenucksi As for the BS444: User Elkin responds in [my blog](https://keptenkurk.wordpress.com/2016/03/30/connecting-the-medisana-bs440-bluetooth-scale-part-5/) that the BS444 is supported. For the other devices cannot tell. 
Anyone?

DjZU, 2017-02-18:
Hi,
I am also owner of the BS444 which I can confirm is supported. 
Actually, I first got the [Sharon LH67301 scale from Leicke](http://www.leicke.eu/fr/products/LH67301) and wanted to read the data from the Linux HTPC since my Android mobile won't pair with these devices. But I have not been successful in trying and sniffing the bluetooth commands (I tried from a live Android on usb but the App was crashing leaving the bt_snoop file empty) so I returned the first scale and got the BS444  (they are almost the same value) to take advantage of keptenkurk work with the BS440.

@lenucksi From a compatible Android device or Live Android on a compatible computer, Enable Bluetooth HCI Snoop Log under Developer Options, then open the scale app, add your scale, make a measurement, when it's successful grab the file /sdcard/btsnoop_hci.log with ES Explorer for instance and post it somewhere so people might help you in understanding the data and adding support to your scale.

lenucksi, 2017-02-18:
I'll see what I can do.

closed 2017-07-01 by keptenkurk

lenucksi, 2017-10-12:
Sorry to revive an old ticket. However I've found the Openscale project https://github.com/oliexdev/openScale , which is creating an open-source app for various BT-LE scales. Among the supported scales is the SBF-70 I mentioned in this ticket. 
There are a few PRs to support it, maybe they could help to add SBF-70 support to this project: https://github.com/oliexdev/openScale/pulls?utf8=%E2%9C%93&q=is%3Apr%20sanitas

keptenkurk, 2017-10-12:
@lenucksi The info here https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothSanitasSbf70.java surely would be of help. From what i quicky saw their protocol is vastly different but that shouldn't be the problem, that is... if you own a SBF-70 ;-)

lenucksi, 2017-10-13:
I do own one of those, how can I be of help with it?

cherocee, 2018-09-02:
Hello,
since I have probelms with a sanitas sbf 70 and my android phone I was looking around if anyone is able to connect to this scale from linux.
may I ask: did anyone get a concretion between linux and the bluetooth scale working?