keptenkurk/BS440 #43: Would BS440.py run on a ESP32 ?

keptenkurk, 2017-02-23:
Knowing that the [ESP32](url) http://esp32.net/supports both Bluetooth and WiFi this coud be a great low power device to interface between the scale and the home network.
The ESP32 is capable of running micropython and, by sheer coicidence [not!],  i do own a [Pycom LoPy](https://www.pycom.io/product/lopy/), but for this purpose cheaper ESP32 boards are available which can be flashed with micropython. From [the docs](https://docs.pycom.io/pycom_esp32/library/network.Bluetooth.html) it's looks like the Bluetooth development has reached a state where is becomes usable for our purpose.

Well then, stop chatting and get on with it!

keptenkurk, 2017-02-26:
First tests show advertisements of the scale but connecting is not possible. Probably because the BTstack (not yet?) supports connecting with the 'random' parameter like in gatttool

keptenkurk, 2017-04-20:
Left a note in the Pycom forum and got that confirmed. Just some more patience.
https://forum.pycom.io/topic/1036/connect-to-ble-device-using-random-adddress-type

elRadix, 2019-10-21:
or you can use ESPHome firmware, I noted that they have managed to get  Xiaomi Mi Scale working on a esp32 board

maybe some inspiration here, I'm not a dev ;-)
https://github.com/esphome/feature-requests/issues/209

keptenkurk, 2019-10-23:
It looks lik BLE support in micropython has improved lately: https://github.com/micropython/micropython/pull/5171
Might be worth another go sometime during the long winternights...

bwynants, 2023-12-26:
> or you can use ESPHome firmware, I noted that they have managed to get Xiaomi Mi Scale working on a esp32 board
> 
> maybe some inspiration here, I'm not a dev ;-) [esphome/feature-requests#209](https://github.com/esphome/feature-requests/issues/209)

https://github.com/bwynants/weegschaal