keptenkurk/BS440 #46: BS440csv: no new graphs

keptenkurk, 2017-02-26:
@DjZU Although data is updated, csv created, user-plot-31.html is updated i keep getting an old test picture i generated before clearing the csv to populate it with real scale data. 
Like render_template does not run...
Can't figure out where to look, any clue?

DjZU, 2017-02-27:
@keptenkurk I switched debug=True to False before uploading the files here and this is the regression. I don't understand why but anyway I switched debug back to True and remove the logging part for BS4440flask.py since it is not useful in production and would get very big with debug=True. I'll make a pull request.

keptenkurk, 2017-02-27:
Thanks a lot! Updates fine now!

closed 2017-02-27 by keptenkurk