keptenkurk/BS440 #50: Don't get Weight and Body info

MarcHeinrich, 2017-03-16:
Hello I am aware it's not a probelm with you current version, but i'm trying to get the data from the bs440.
The Problem is i do only get the userinformation which is characteristic "8a82" the other ones("8a21" and "8a22") i do not get any data from , also i do not get a 0x25 before the 0x84 from my user data it would be nice if you have any idea why this could be.
I looked at the code and you wrote "the scale will emit 30 Indications on 0x1b and 0x1e each" i dont get that where do i read these?

MarcHeinrich, 2017-03-16:
Update:
I do get all information now i realized i needed to send a timestamp each time i want to read a dataset

closed 2017-03-16 by MarcHeinrich

keptenkurk, 2017-03-16:
Thanks for reporting, @MarcHeinrich ! Yes, by sending the unix timestamp the scale syncs it's RTC.