keptenkurk/BS440 #54: Ohter Scales

MarcHeinrich, 2017-03-27:
Hello i know it's not about the BS440/BS430 or BS444 which work with this projekt, but i tried it with some oder scales from different companys and most of them seem to work with the same characteristics so i thought it could be possible to get the data from them too but i do not get the measurement data.
I believe thats because they need some kind of authentication (like the 0x23,0x02 before the time stamp for the medisana scales or maybe something more complicated)
Do you have any info on that or a tipp on how to tackle this problem?

DjZU, 2017-03-27:
@MarcHeinrich From an Android device with the scale manufacturer app running, you need to grab the HCI Snoop Log. Please see #40

MarcHeinrich, 2017-03-27:
Sry if i opened this issue unnecessarily my thought was the other issue was only for other Medisana scales. 
Anyway thank you for your answer.
I will try to look at it in the next days hopefully i can see something in this log

MarcHeinrich, 2017-04-05:
Hello again 
I looked at some logs from the communication between the original Fitvigo app and the scale. I was able to extract the messages they send each other but one of them seems so be some kind of authentication key which i do not unterstand how it is put together 

This is the process i got from the log: (":" is used here to divide the hex value so its easier to read )
all data is send to characteristic 8a81 which is download information or command 
and i receive all data from 8a82 which is upload information or command

check characteristics
subscribe indications
get: a0:b2:da:1a:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00  //line 3241
ack : 1e
send: 21:e6:6d:5f:59
ackw: 13
get: a1:01:00:a6:4f:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
ack : 1e
send: 20:b3:da:bc:b0	//thats the one that changes from an other scan a few days later i got 20:b3:da:bd:5d
ackw: 13
//this part is repeated for every profile until profil 9
bekomme: 83:01:4d:61:72:63:20:20:20:20:20:20:20:20:20:20:20:20:00:00 (name of the profile in the current slot beginning at 4d)
ack : 1e
//
send: 03:01:4d:61:72:63:20:20:20:20:20:20:20:20:20:20:20:20:20:20 (active profile in the app here Marc)
ackw: 13
send: 51:07:01:01:1b:d7:d6
ackw: 13
send: 02:24:4f:a6:0d
ackw: 13
send: 22
ackw: 13

//from here its like round 2
check characteristics
subscribe indications
get: a1:01:00:a6:4f:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
ack : 1e
send: 20:b3:da:bc:b0		//thats the one that changes from an other scan a few days later i got 20:b3:da:bd:5c
ackw: 13
send: 02:2e:4f:a6:0d
ackw: 13
get: c0:17:01:01:1b:d7:d6:00:00:00:00:00:00:00:00:00:00:00:00:00
ack: 1e
send: 51:07:01:01:1b:d7:d6
ackw: 13
send: 22
ackw: 13

Any ideas on how to know how these are made i tried some timestamps but can't seem to fit any
I get the feeling it could be a generated key which is used for pairing but if thats the case i dont know how i could try to know how its generated
Do see anything i missed?

Thanks in advance

MarcHeinrich, 2017-04-27:
Update: I know for most of the communication how its done but i cant get behind 1 byte it looks like this byte is importen because i cant get a working connection without it.
The questionable byte is in the message which always starts with 20 
20:b3:da:bc:b0 in this one it would be bc it decreases 1 each day and randomly counts up some amount most of the time 7 but i hat a jump from 07 to 7b

Any idea how this works?

Dylantje, 2017-06-02:
perhaps someone have this working with a whiting body scale?
Looks almost the same....

Or perhaps is there a place in the script where i can change something..

Hope...

Thanks

MarcHeinrich, 2017-06-06:
Hallo
I dont't know about the whithings scale since it is a wi-fi only scale from my information and i need bluetooth le but if it shares the same protocoll it could be a worthwhile addition
do you have a link to the whitings protocol?

For the ADE Scale I do have the missing byte the calculation was strange though.
The nasty thing is the measurements are only saved on the scale when it remembers a user with the same weight (or close to it) and i don't know yet how to set a user. But progress has been made :D

Which Skript do you want to change?

Tristan79, 2017-06-07:
Whithings has some
time ago changed from http protocol to https encryption on its wifi network connection, forcing to either use their cloud :-/ or  can try to decrypt it with a man in the middle attack, attacking the ssl... to read it out... you need something like...

https://mitmproxy.org/

i traded it in for a bs440, when i found this github page :-)

MarcHeinrich, 2017-06-07:
@Tristan79 Yeah they forcing us to use their cloud or something like it is the main reason i am doing this.
Thats tough with the ssl but i may have a look sometime but like i said at the moment i am focusing on bluetooth le and i am happy to be done with my scale today :D

The good thing is a bunch of devices use the protocoll with very litttle difference what makes it easier to find out how it works.

The funny thing i encounterd a while ago is Philips uses the same technique as ADE but are much more expensive. .It's literally the same with the exception of a password key which is send during the first connection anyway xD

closed 2017-06-19 by MarcHeinrich