keptenkurk/BS440 #72: BS440google broken?

jovandeginste, 2017-07-05:
I tried to get the `BS440google` plugin working, and while I had quite some work getting it to start (making it a `Plugin`), it errored out when trying to push the data to google fit. Anyone still maintaining this plugin?

jovandeginste, 2017-07-05:
ping @managementboy

keptenkurk, 2017-07-06:
actually just had it working once, but i remember that the authentication process was quite a pain...

jovandeginste, 2017-07-06:
Yes I can update the wiki to get to the point of getting the json, but since I didn't get the final part working so maybe I shouldn't touch the wiki just yet ;-)

thomashommel, 2019-10-20:
I'm stuck at the last part, I have generated the json file (google.json) and it's copied to the BS440 folder on my pi. But what do I do then?

I don't understand this part: _Copy and rename this file to the BS440 directory on the Pi and refer to this file in BS440.ini in the user section for the user this account belongs to._

Can you please help? @jovandeginste @keptenkurk @managementboy

keptenkurk, 2019-10-21:
Thomas, 
The issue is that this plugin was never rewritten into the Plugin structure,
At some point i decided to create a structure where 3rd party plugins could be added wiithout the need to change the basic communication structure. Plugins process the received data into CSV, Domoticz, Email etc just to name a few...
The google integration was never converted into this new structure so it will not work anymore.
I started to fix this (as i'm a Gioogle Fit user now) so i hope nothing dramatically has changed over the years in accessing Google Fit by an application...
Stay tuned and hope for lots of rain (and little spouse priority requests)

thomashommel, 2019-10-21:
Perfect! 

I indeed spotted that one and added an BS440google.ini file and added "class Plugin" in the BS440google.py file, which made it load the plugin, but at the plugin execute phase the code would throw me an error and since I'm not a developer, that is where I got stuck :)

Would love to have this integration! Keep up the good work!

keptenkurk, 2019-10-23:
Ok, here you go! Just pushed the updates into master. Tested by using BS440test.py and seems to be working fine. Also see updated Wiki pages on this plugin.

keptenkurk, 2019-10-24:
Hmmm. Here the plugin causes the complete script to terminate silently after loading... (whereas BS440test loads it exactly the same way)... need to investigate!

thomashommel, 2019-10-24:
This is the error I get back:

Traceback (most recent call last):
  File "BS440.py", line 318, in <module>
    plugin.execute(config, persondata, weightdatasorted, bodydatasorted)
  File "plugins/BS440google.py", line 161, in execute
    self.AddGoogle(self.googleClient, weightdata[0]['weight'], WEIGHTD, googleauthfile)
  File "plugins/BS440google.py", line 117, in AddGoogle
    value=[dict(fpVal=value)],), ],
  File "/usr/local/lib/python2.7/dist-packages/googleapiclient/_helpers.py", line 130, in positional_wrapper
    return wrapped(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/googleapiclient/http.py", line 856, in execute
    raise HttpError(resp, content, uri=self.uri)
googleapiclient.errors.HttpError: <HttpError 400 when requesting https://www.googleapis.com/fitness/v1/users/me/dataSources/raw%3Acom.google.weight%3A<MYNUMBERS>%3Aunknown%3Aunknown%3A10000001/datasets/<DATASET1>-<DATASET2>?alt=json returned "Unable to fetch DataSource for Dataset: raw:com.google.weight:<MYNUMBERS>:unknown:unknown:10000001">

The <MYNUMBERS> part is the same and a set of numbers, but on the requested URL it is preceded by 3A, which seems like it doesn't belong there.

Thank you for your help so far!

keptenkurk, 2019-10-24:
Not sure what's going on here. Some difference comparing your responses with mine: 
The 401 error (authentication error) in: Thu, 24 Oct 2019 09:31:49 INFO new_request Refreshing due to a 401 (attempt 1/2).
The %3A is not the problem. It is just the Hex presentation of the colon (:) 

Are you sure creating the authentication file worked ok? Does your application show in the Google Fit app under Settings | Connected Apps  (BS440 is listed in mine) and shows to have read/write access.

keptenkurk, 2019-10-25:
I discovered a bit of a design issue. Not all users in my household have a Google Fit account but the use of this plugin is defined globally when the application starts. Must think of a way to disable the use of a plugin for one/some of the users within the plugin .ini

keptenkurk, 2019-10-25:
Pushed an update taking care of the issue mentioned above

thomashommel, 2019-10-27:
Thank you so much for your help. I still receive the following error:

Sun, 27 Oct 2019 20:56:31 INFO     GetGoogleClient Google client created
Sun, 27 Oct 2019 20:56:31 INFO     AddGoogle Created a new dataset: "DATASET NUMBERS"
Sun, 27 Oct 2019 20:56:31 INFO     method URL being requested: GET https://www.googleapis.com/fitness/v1/users/me/dataSources/raw%3Acom.google.weight%3A"DATASOURCE NUMBERS"%3AMedisana%3ABS430%3A10000001?alt=json
Sun, 27 Oct 2019 20:56:32 INFO     AddGoogle Data Source ID: raw:com.google.weight:"DATASOURCE NUMBERS":Medisana:BS430:10000001
Sun, 27 Oct 2019 20:56:32 INFO     method URL being requested: PATCH https://www.googleapis.com/fitness/v1/users/me/dataSources/raw%3Acom.google.weight%3A"DATASOURCE NUMBERS"%3AMedisana%3ABS430%3A10000001/datasets/"DATASET NUMBERS"?alt=json
googleapiclient.errors.HttpError: <HttpError 400 when requesting https://www.googleapis.com/fitness/v1/users/me/dataSources/raw%3Acom.google.weight%3A"DATASOURCE NUMBERS"%3AMedisana%3ABS430%3A10000001/datasets/"DATASET NUMBERS"?alt=json returned "Unable to fetch DataSource for Dataset: raw:com.google.weight:"DATASOURCE NUMBERS":Medisana:BS430:10000001">

It seems like the dataset is not being created correctly? Although the log says AddGoogle did create it. I tried to check if I could find the dataset with the google API explorer. This is the error if I test Users.dataSources.datasets: patch:

{
 "error": {
  "errors": [
   {
    "domain": "global",
    "reason": "invalidArgument",
    "message": "DataSourceId in request: raw:com.google.weight:"DATASOURCE NUMBERS":Medisana:BS430:10000001 does not match DataSourceId in Dataset. "
   }
  ],
  "code": 400,
  "message": "DataSourceId in request: raw:com.google.weight:"DATASOURCE NUMBERS":Medisana:BS430:10000001 does not match DataSourceId in Dataset. "
 }
}

thomashommel, 2019-10-27:
Intersting, if I try the Users.dataSources.datasets: get on the Google API explorer I do get the confirmation it exists, although it is empty:

{
 "minStartTimeNs": "NUMBERS",
 "maxEndTimeNs": "NUMBERS",
 "dataSourceId": "raw:com.google.weight:"DATASOURCE NUMBERS":Medisana:BS430:10000001",
 "point": []
}

keptenkurk, 2020-06-04:
Seems nothing going on anymore.

closed 2020-06-04 by keptenkurk