keptenkurk/BS440 #78: Connect to BS445

julianvollmer, 2017-10-11:
Hi keptenkurk, 

first I wan't to thank you for your great documented work.
I wan't to connect my BS445 to a Raspberry Pi 3, I think I've configured and installed everything. 
I can connect via gattool:

```bash
pi@raspberrypi:~ $ sudo gatttool -t random -b C1:D9:F8:90:29:1C -I
[C1:D9:F8:90:29:1C][LE]> connect
Attempting to connect to C1:D9:F8:90:29:1C
Connection successful
[C1:D9:F8:90:29:1C][LE]> 
[C1:D9:F8:90:29:1C][LE]> exit
```

When I monitor the log file while i'am running the python script, I get the following warning and nothing else happens.

```bash
Wed, 11 Oct 2017 16:40:14 INFO     scan Starting BLE scan
Wed, 11 Oct 2017 16:40:24 INFO     scan Discovered 70:56:81:E9:85:FA (None)
Wed, 11 Oct 2017 16:40:24 INFO     scan Discovered 10:96:24:74:61:A2 (None)
Wed, 11 Oct 2017 16:40:24 INFO     scan Discovered C1:D9:F8:90:29:1C (0131971C2990F8D9C1)
Wed, 11 Oct 2017 16:40:24 INFO     scan Found 3 BLE devices
Wed, 11 Oct 2017 16:40:24 INFO     connect Connecting to C1:D9:F8:90:29:1C with timeout=8
Wed, 11 Oct 2017 16:40:25 DEBUG    get_handle Looking up handle for characteristic 00008a22-0000-1000-8000-00805f9b34fb
Wed, 11 Oct 2017 16:40:26 DEBUG    _save_charecteristic_callback Found characteristic 00002a04-0000-1000-8000-00805f9b34fb, value handle: 0x7
Wed, 11 Oct 2017 16:40:26 DEBUG    _save_charecteristic_callback Found characteristic 00008a82-0000-1000-8000-00805f9b34fb, value handle: 0x25
Wed, 11 Oct 2017 16:40:27 WARNING  get_handle No characteristic found matching 00008a22-0000-1000-8000-00805f9b34fb
Traceback (most recent call last):
  File "BS440.py", line 263, in <module>
    handle_body = device.get_handle(Char_body)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/device.py", line 208, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: No characteristic found matching 00008a22-0000-1000-8000-00805f9b34fb```

Can you help me to find the right characteristic?

Thank you, 

Julian

keptenkurk, 2017-10-11:
@julianvollmer It seems that the BS445 uses different characteristics for its BLE communications. I'm afraid you'll need to download the app, enable BT Snooping [which i've never done myself] and have a look what's going on and report back so we can code around it... BS430 an BS440 behave the same and BS410 had  some minor differences, but this BS445?

julianvollmer, 2017-10-13:
Thanks you for offering support.
I've an Android Device now and will give Bluetooth Snooping a try.

julianvollmer, 2017-10-16:
I've done some BT Snooping, here the file exported with Wireshark to JSON-Format.
I think all relevant parts containing the attribute and value: 
`"frame.protocols": "bluetooth:hci_h4:bthci_acl:btl2cap:btatt",`
Do you think this helps?

Best Regards,

Julian

[bt_medisana.log](https://github.com/keptenkurk/BS440/files/1387125/bt_medisana.log)

keptenkurk, 2017-10-17:
@julianvollmer I think that helps. 
Give me a few days as i'm in the middle of a bedroom renovation - painting scheduled by the CEO for tonight.
Meanwhile: could you also create/upload the .pcap file? Using wireshark makes things easier to read. Not having any luck in opening the file above in Wireshark (v2.2.1)

julianvollmer, 2017-10-17:
Great, next days/weeks will be reserved to learn something about Bluetooth and BT Low Energy. 
Here is also a bluetooth thermostat that will be connected to my MagicMirror.

I can't upload .pacp-files, so here is the source file from my phone. 
I can open it with Wireshark (2.4.2).
I think this should work for you too.
[bt_medisana_wireshark.log](https://github.com/keptenkurk/BS440/files/1391383/bt_medisana_wireshark.log)

Good Luck with bedroom renovation and your CEO  👍

mod25, 2018-01-23:
keptenkurk
i have the same issue with my bs444. what can i do to assist?

Best

DjZU, 2018-01-24:
@mod25 I am also the owner of the BS444 which works out of the box with this script. Please open a new issue with your log if you get in trouble.
@julianvollmer Could you extract the required information from the snooped log and make it work?

julianvollmer, 2018-01-24:
The Projekt is standing still for a while and at the moment there is no much time.
I'am a father of a new born child, so priorities have changed ;-)
Extracting the required information from the snooped log and make it work seems to be to much work.
If there are some small concrete questions,  I can answer, no problem. But going deep in BLE is to much work. Sorry, but as you see I am always on and trying to help.
Best,
Julian

mod25, 2018-01-24:
@DjZU will do but my problem has the same erros as julianvollmer has so i thought it is something else

DjZU, 2018-01-24:
@julianvollmer Please get the BS445 closer to your BT receiver and try again. I mean very close for testing like less than 1 meter.

julianvollmer, 2018-01-24:
Which steps should I do? Uploading my Data? Connecting?

DjZU, 2018-01-24:
1. Get your BS445 as close as possible of your bluetooth receiver
2. Remove your shoes and socks ^^
3. Step on the scale
4. Check your BS440.log

DjZU, 2018-01-24:
FYI, I had a look at your snooped log file with Wireshark and the handles seem to be the same ones.

julianvollmer, 2018-01-24:
Here we go...

[btsnoop_hci.log](https://github.com/keptenkurk/BS440/files/1661210/btsnoop_hci.log)

DjZU, 2018-01-24:
I meant the log of this script like in your first post not the snoop

julianvollmer, 2018-01-28:
`pi@magicMirror:~/workspace/BS440 $ sudo btmgmt le on
hci0 Set Low Energy complete, settings: bondable ssp br/edr le secure-conn 
pi@magicMirror:~/workspace/BS440 $ sudo python BS440.py 
2018-01-28 11:05:52,573 INFO     <module> No plugins configured.
2018-01-28 11:05:52,574 INFO     <module> BS440 Started
2018-01-28 11:05:52,620 INFO     init_ble_mode hci0 Set Low Energy complete, settings: bondable ssp br/edr le secure-conn 

Traceback (most recent call last):
  File "BS440.py", line 263, in <module>
    handle_body = device.get_handle(Char_body)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/device.py", line 208, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: No characteristic found matching 00008a22-0000-1000-8000-00805f9b34fb
pi@magicMirror:~/workspace/BS440 $ `

DjZU, 2018-01-29:
Please check issue #80 and check your PyGATT version
`pip list | grep pygatt`

julianvollmer, 2018-01-29:
Thx, works fine!