keptenkurk/BS440 #80: BS444 doesn't work

mod25, 2018-01-24:
Hi everone,

first of all really great work thank you for this.

everytime i us BS440.py with mail plugin i get following log:
Wed, 24 Jan 2018 10:28:11 INFO     scan Discovered E8:2A:55:47:FA:F2 (013197F2FA47552AE8)
Wed, 24 Jan 2018 10:28:11 INFO     scan Found 6 BLE devices
Wed, 24 Jan 2018 10:28:11 INFO     connect Connecting to E8:2A:55:47:FA:F2 with timeout=8
Wed, 24 Jan 2018 10:28:19 ERROR    connect Timed out connecting to E8:2A:55:47:FA:F2 after 8 seconds.
Wed, 24 Jan 2018 10:28:19 INFO     connect Connecting to E8:2A:55:47:FA:F2 with timeout=8
Wed, 24 Jan 2018 10:28:19 DEBUG    get_handle Looking up handle for characteristic 00008a22-0000-1000-8000-00805f9b34fb
Wed, 24 Jan 2018 10:28:21 DEBUG    _save_charecteristic_callback Found characteristic 00002a04-0000-1000-8000-00805f9b34fb, value handle: 0x7
Wed, 24 Jan 2018 10:28:21 DEBUG    _save_charecteristic_callback Found characteristic 00008a82-0000-1000-8000-00805f9b34fb, value handle: 0x25
Wed, 24 Jan 2018 10:28:22 WARNING  get_handle No characteristic found matching 00008a22-0000-1000-8000-00805f9b34fb

If needed i would upload btsnooping log.
thank you

mod25, 2018-01-24:
[mod25.pcap.zip](https://github.com/keptenkurk/BS440/files/1659286/mod25.pcap.zip)

DjZU, 2018-01-24:
Please get the BS444 closer to your BT receiver and try again. I mean very close for testing like less than 1 meter.

mod25, 2018-01-24:
done. But with no changes. same error:

pi@home:~/BS440 $ python BS440.py 
2018-01-24 21:42:34,927 INFO     <module> Configured plugins: BS440mail
2018-01-24 21:42:34,928 INFO     <module> Loading plugin: BS440mail
2018-01-24 21:42:35,040 INFO     <module> All plugins loaded.
2018-01-24 21:42:35,041 INFO     <module> BS440 Started
2018-01-24 21:42:35,163 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn 

Traceback (most recent call last):
  File "BS440.py", line 263, in <module>
    handle_body = device.get_handle(Char_body)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/device.py", line 208, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: No characteristic found matching 00008a22-0000-1000-8000-00805f9b34fb
pi@home:~/BS440 $ cat BS440.log 
Wed, 24 Jan 2018 21:42:34 INFO     <module> Configured plugins: BS440mail
Wed, 24 Jan 2018 21:42:34 INFO     <module> Loading plugin: BS440mail
Wed, 24 Jan 2018 21:42:35 INFO     <module> All plugins loaded.
Wed, 24 Jan 2018 21:42:35 INFO     <module> BS440 Started
Wed, 24 Jan 2018 21:42:35 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn 

Wed, 24 Jan 2018 21:45:49 DEBUG    start gatttool_cmd=gatttool -i hci0 -I
Wed, 24 Jan 2018 21:45:49 INFO     run Running...
Wed, 24 Jan 2018 21:45:49 INFO     scan Starting BLE scan
Wed, 24 Jan 2018 21:46:30 INFO     scan Discovered E8:2A:55:47:FA:F2 (013197F2FA47552AE8)
Wed, 24 Jan 2018 21:46:30 INFO     scan Found 6 BLE devices
Wed, 24 Jan 2018 21:46:30 INFO     connect Connecting to E8:2A:55:47:FA:F2 with timeout=8
Wed, 24 Jan 2018 21:46:30 DEBUG    get_handle Looking up handle for characteristic 00008a22-0000-1000-8000-00805f9b34fb
Wed, 24 Jan 2018 21:46:31 DEBUG    _save_charecteristic_callback Found characteristic 00002a27-0000-1000-8000-00805f9b34fb, value handle: 0xe
Wed, 24 Jan 2018 21:46:31 DEBUG    _save_charecteristic_callback Found characteristic 00008a82-0000-1000-8000-00805f9b34fb, value handle: 0x25
Wed, 24 Jan 2018 21:46:32 WARNING  get_handle No characteristic found matching 00008a22-0000-1000-8000-00805f9b34fb

DjZU, 2018-01-25:
You're missing handles that's why I thought about distance but it might be a bug in an outdated depency. The log should include information about that.
Can you check your system and dependencies versions?

keptenkurk, 2018-01-25:
@mod25 What version of Pygatt do you use? I've seen people having trouble reading characteristics using Pygatt 3.2 (where 3.1.1 was working fine). See https://github.com/peplin/pygatt/issues/152, https://github.com/peplin/pygatt/issues/161 and https://github.com/peplin/pygatt/issues/166

mod25, 2018-01-26:
works.
sudo apt-get install python-dev (for coverage.py)
sudo pip install "pygatt[GATTTOOL]==3.1.1"
thank you

keptenkurk, 2018-01-26:
Great! Will add a comment in the readme for the time being.

closed 2018-01-26 by keptenkurk