keptenkurk/BS440 #91: Can't connect BS444 bluetooth

Caroga, 2020-09-02:
Hi there!
I've been trying to connect the medisana scale and I keep getting "timeouts" while testing the Bluetooth connecting. I was hoping you might be able to give more insight into debugging this as I am currently at a loss.

I've installed Bluez by compiling the latest binaries: bluez-5.54
I'm using it on an RPI zero W.

I can find the Bluetooth mac address but connecting to it will eventually timeout and return a `Error: connect error: Connection refused (111)`. 
You've stated that this probably would be because of not running in LE mode but it is. If I try to connect to it with LE off it will disconnect immediately.

Do you have any pointers or do you know how to debug this? Any help would be greatly appreciated.
Bedankt!

keptenkurk, 2020-09-02:
Hi @Caroga I will be away from my PI for the next week and a half. Will try to get an answer by then.

Caroga, 2020-09-02:
No worries, I'm very glad you are willing to help.
Also, I found some older issues "kinda" addressing the same thing so I'll be going through that too.

Caroga, 2020-09-24:
Hi there, I will be trying to connect this setup again tomorrow, are you available to help me with some debugging?

keptenkurk, 2020-09-24:
I was just looking into it. Are you trying to connect to the scale manually (with gatttool) ?  If so: Make sure that the scale is in the state where it is accepting connections (just after weighing and when values are stored, see youtube). I also ran into this page: https://crownstone.rocks/nl/2014/07/23/linux-and-ble where they set the adapter to ignore it's own address by 
sudo hcitool -i hci1 lecc --random [MAC address]  
Or are you running the BS440.py and getting the error in the debug log..?

Caroga, 2020-09-24:
Thank you for your time!
What video are you referencing? I did make sure that when using gattool I was doing so after weighing. I didn't know if I had to first register the device with the app on my phone so I've done this but haven't gotten the time to check again. I will do so this morning.

The error I get is from using gattool, but when running BS440.py I remember it stating that it was searching, connecting but then failed and reset after a short while. I will post this tomorrow again.

keptenkurk, 2020-09-25:
The one in the first part of the blog. But  i now realise that you need to step up on the scale to even be able to see it in a scan...

keptenkurk, 2020-09-25:
https://youtu.be/2Px-D8Ye_u0

Caroga, 2020-09-26:
So I just finally got time to get to this. This is the logs from BS440. 

```
Sat, 26 Sep 2020 19:59:56 INFO     <module> BS440 Started
Sat, 26 Sep 2020 19:59:56 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered ssp br/edr le secure-conn

Sat, 26 Sep 2020 19:59:57 DEBUG    start gatttool_cmd=gatttool -i hci0 -I
Sat, 26 Sep 2020 19:59:58 INFO     scan Starting BLE scan
Sat, 26 Sep 2020 19:59:58 INFO     run Running...
Sat, 26 Sep 2020 20:00:08 INFO     scan Discovered 00:09:B0:36:62:60 (None)
Sat, 26 Sep 2020 20:00:08 INFO     scan Discovered 5D:DD:DD:FE:E5:A0 (None)
Sat, 26 Sep 2020 20:00:08 INFO     scan Discovered 56:9D:F8:60:64:0C (None)
Sat, 26 Sep 2020 20:00:08 INFO     scan Discovered 5E:42:8C:61:67:8D (None)
Sat, 26 Sep 2020 20:00:08 INFO     scan Discovered 19:DD:6F:F8:1F:AF (None)
Sat, 26 Sep 2020 20:00:08 INFO     scan Discovered 4E:69:63:3B:9C:6C (None)
Sat, 26 Sep 2020 20:00:08 INFO     scan Found 6 BLE devices
Sat, 26 Sep 2020 20:00:08 INFO     scan Starting BLE scan
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered 00:09:B0:36:62:60 (None)
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered 56:9D:F8:60:64:0C (None)
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered 5D:DD:DD:FE:E5:A0 (None)
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered A4:C1:38:3C:43:0E (0131980E433C38C1A4)
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered 5E:42:8C:61:67:8D (None)
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered 4E:69:63:3B:9C:6C (None)
Sat, 26 Sep 2020 20:00:18 INFO     scan Discovered 19:DD:6F:F8:1F:AF (None)
Sat, 26 Sep 2020 20:00:18 INFO     scan Found 7 BLE devices
Sat, 26 Sep 2020 20:00:18 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=5
Sat, 26 Sep 2020 20:00:23 ERROR    connect Timed out connecting to A4:C1:38:3C:43:0E after 5 seconds.
Sat, 26 Sep 2020 20:00:23 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=5
Sat, 26 Sep 2020 20:00:28 ERROR    connect Timed out connecting to A4:C1:38:3C:43:0E after 5 seconds.
Sat, 26 Sep 2020 20:00:28 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=5
Sat, 26 Sep 2020 20:00:33 ERROR    connect Timed out connecting to A4:C1:38:3C:43:0E after 5 seconds.
Sat, 26 Sep 2020 20:00:33 INFO     scan Starting BLE scan
Sat, 26 Sep 2020 20:00:33 ERROR    scan BLE adapter requires reset after a scan as root- call adapter.reset()
Sat, 26 Sep 2020 20:00:40 INFO     scan Starting BLE scan
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered 5D:DD:DD:FE:E5:A0 (None)
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered A4:C1:38:3C:43:0E (0131980E433C38C1A4)
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered 00:09:B0:36:62:60 (None)
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered 56:9D:F8:60:64:0C (None)
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered 5E:42:8C:61:67:8D (None)
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered 4E:69:63:3B:9C:6C (None)
Sat, 26 Sep 2020 20:00:50 INFO     scan Discovered 19:DD:6F:F8:1F:AF (None)
Sat, 26 Sep 2020 20:00:50 INFO     scan Found 7 BLE devices
```

The above is the MAC address from the device as found by the `hcitool lescan`
```
A4:C1:38:3C:43:0E 0131980E433C38C1A4
A4:C1:38:3C:43:0E (unknown)
```

I keep checking my device and it does show a Bluetooth icon but very short and not as steady as shown on your video. Syncing with my phone does seem to work though. 

I copied a segment from the log that shows the inability to connect but in case it would be helpful I could post more.

keptenkurk, 2020-09-27:
@Caroga Two questions pop up when reading your log:
* What version of BS440.py are you using? The default timeout for connecting devices is set to 8 seconds since the commits following the very informative thread in #56  Your log shows a timeout of 5 seconds which seems to be a bit too short in the experience of BS444 users.
* Have you set the scale type to BS444 in the BS440.ini file? This changes the addresstype to "random" (as opposed to "public") and handles the difference in timestamps between scale versions. 

Then a general advise is to have the RPI zero very close to the scale (1m max). The scale will be detected over a longer distance but connecting to it requires it to be very close.
From #80 : be sure to not use latest pygatt but stick to 3.1.1 if possible (never checked if later ones do work...).

If this still doesn't give any joy we'll need to remove some exception lines in the code temporarily and get some more details of what's happening.

Caroga, 2020-09-27:
Hi @keptenkurk , let me respond in order:

- I have downloaded BS440-2.0 from github, I've double checked my history which shows `wget https://github.com/keptenkurk/BS440/archive/2.0.zip`
- I did not change any scale types, could you explain to me what configuration this should be and where? Would it be `type: BS444 ` in the scale segment? 

Below is a cat of the configuration file:

```
$ cat BS440.ini

# BS440.ini
# Initialization file for BS440.py
# Change to your preference
# Store this file in the working folder of BS440.py

# Scale properties
[Scale]
ble_address: A4:C1:38:3C:43:0E
device_name: 0131980E433C38C1A4

# Program behaviour
# Loglevel: debug | info | critical | error
[Program]
loglevel: debug
logfile: BS440.log
```

I checked pygatt and I have version 4.x installed, I will downgrade this after fixing the scale type setting and confirming I have the right version installed of BS440. This way I can check for you if the newest version is working or not.

I think I will download the git repository now and try again.

Caroga, 2020-09-27:
After posting I checked: the [2.0 release](https://github.com/keptenkurk/BS440/releases/tag/2.0) does not contain the same contents as the current master, the difference is in the configuration file. I will download the repository version now.

Caroga, 2020-09-27:
I've downloaded the [repository](https://github.com/keptenkurk/BS440/archive/master.zip) and configured the following .ini file:

```
r# cat BS440.ini
# BS440.ini
# Initialization file for BS440.py
# Change to your preference
# Store this file in the working folder of BS440.py

# Scale properties
# Model: BS410, BS430, BS440, BS444
[Scale]
ble_address: A4:C1:38:3C:43:0E
device_name: 0131980E433C38C1A4
device_model: BS444

# Program behaviour
# Loglevel: debug | info | critical | error
[Program]
loglevel: debug
logfile: BS440.log
# Enable your plugins here, use the comma separated basename from the plugins/
# Example:
# plugins: BS440csv, BS440mail
```

I've also downgraded pygatt:
```
# pip show pygatt
Name: pygatt
Version: 3.1.1
Summary: Python Bluetooth LE (Low Energy) and GATT Library
Home-page: https://github.com/peplin/pygatt
Author: Chris Peplin <github@rhubarbtech.com>
Author-email: github@rhubarbtech.com
License: Apache 2.0 and MIT
Location: /usr/local/lib/python2.7/dist-packages
Requires: pyserial, enum34
Required-by:
```

I've done a testrun but am unable to connect. 
```
Sun, 27 Sep 2020 11:37:57 INFO     <module> No plugins configured.
Sun, 27 Sep 2020 11:37:57 INFO     <module> BS440 Started
Sun, 27 Sep 2020 11:37:57 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered ssp br/edr le secure-conn
Sun, 27 Sep 2020 11:37:59 DEBUG    start gatttool_cmd=gatttool -i hci0 -I
Sun, 27 Sep 2020 11:37:59 INFO     scan Starting BLE scan
Sun, 27 Sep 2020 11:37:59 INFO     run Running...
Sun, 27 Sep 2020 11:38:09 INFO     scan Discovered 00:09:B0:36:62:60 (None)
Sun, 27 Sep 2020 11:38:09 INFO     scan Discovered 77:8F:D3:F1:84:8A (None)
Sun, 27 Sep 2020 11:38:09 INFO     scan Discovered 0E:A4:DB:0B:13:0A (None)
Sun, 27 Sep 2020 11:38:09 INFO     scan Discovered 71:B1:B1:36:0E:F5 (None)
Sun, 27 Sep 2020 11:38:09 INFO     scan Discovered 78:BA:6B:F7:45:FD (None)
Sun, 27 Sep 2020 11:38:09 INFO     scan Discovered 43:86:BC:4D:6B:C0 (None)
Sun, 27 Sep 2020 11:38:09 INFO     scan Found 6 BLE devices
Sun, 27 Sep 2020 11:38:09 INFO     scan Starting BLE scan
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered 0E:A4:DB:0B:13:0A (None)
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered A4:C1:38:3C:43:0E (0131980E433C38C1A4)
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered 00:09:B0:36:62:60 (None)
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered 78:BA:6B:F7:45:FD (None)
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered 71:B1:B1:36:0E:F5 (None)
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered 77:8F:D3:F1:84:8A (None)
Sun, 27 Sep 2020 11:38:19 INFO     scan Discovered 43:86:BC:4D:6B:C0 (None)
Sun, 27 Sep 2020 11:38:19 INFO     scan Found 7 BLE devices
Sun, 27 Sep 2020 11:38:19 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=8
Sun, 27 Sep 2020 11:38:27 ERROR    connect Timed out connecting to A4:C1:38:3C:43:0E after 8 seconds.
Sun, 27 Sep 2020 11:38:27 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=8
Sun, 27 Sep 2020 11:38:35 ERROR    connect Timed out connecting to A4:C1:38:3C:43:0E after 8 seconds.
Sun, 27 Sep 2020 11:38:35 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=8
Sun, 27 Sep 2020 11:38:43 ERROR    connect Timed out connecting to A4:C1:38:3C:43:0E after 8 seconds.
Sun, 27 Sep 2020 11:38:43 INFO     scan Starting BLE scan
Sun, 27 Sep 2020 11:38:44 ERROR    scan BLE adapter requires reset after a scan as root- call adapter.reset()
Sun, 27 Sep 2020 11:38:50 INFO     scan Starting BLE scan
```

What lines should I enable for more debugging?

keptenkurk, 2020-09-27:
Ok let's see. Connecting is done by pygatt calling "gatttool" with the proper commandline parameters. Gatttool is executed  through "pexpect" which fires the command and waits for the response. This response should fit one of the predefined patterns. In case of connecting it will wait for the a response from gatttool which exactly matches the regular expression r'Connection successful.*\[LE\]>'
So far i failed to get the actual response of the gatttool command visible (which will worst case show error 111, or if we're lucky a Connect message which doesn't match the pattern). 
Hopefully some of the other BS444 users will bump in with some advice...

keptenkurk, 2020-09-27:
I've never used the Phone app but think that using that would interfere...

awillems, 2020-09-30:
Yes, definitely, If the app is open on the phone, the scale will connect to the phone faster then the rpi.  And the rpi will be unable to connect.  I cannot actually speak to much to the BS440 though I am using "Etekcity Digital Body" https://www.amazon.com/gp/product/B07D7R25JV/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
had to modify a decent amount.  But when I was testing with gatttool I had some issues.  This might only apply to my scale but here are some thing to try maybe.

1. close app on phone make sure its not connecting to your phones BL maybe turn BL off on your phone for testing
2.  try rebooting your pi
3. a lot of people have suggested using "-t random" flag if having connecting issues, when connecting with "sudo gatttool -t random -b xx:xx:xx:xx:xx:xx -I"  I did this at the start and was never able to connect but the minute I removed the  "-t random" flag it connected that might be an issue not sure.

Caroga, 2020-09-30:
Thank you @awillems, by removing the `-t random` from the gattool command I was able to connect to the device by using `gatttool -b A4:C1:38:3C:43:0E -I` and typing in `connect`.

```
# gatttool -b A4:C1:38:3C:43:0E -I 
[A4:C1:38:3C:43:0E][LE]> connect
Attempting to connect to A4:C1:38:3C:43:0E
Connection successful
[A4:C1:38:3C:43:0E][LE]> 

```

That's already a good step forward! 

Did you have to make any other additions to the script for it to connect by BS440.py?

keptenkurk, 2020-10-01:
@awillems  Thanks for bumping in! 
To change connecting to public you could change line 242:
    addresstype = pygatt.BLEAddressType.random 
into
    addresstype = pygatt.BLEAddressType.public 
(if the -t option is not set gatttool will take public as the default addresstype)

Caroga, 2020-10-01:
Thanks to this addition and some extensive testing I got the setup to work. I even got the MQTT plugin working.
The log file does state some messages like `Unreliable data received. Unable to process` during the process. 
I don't know what to make from this but the software does work eventually.
```
Thu, 01 Oct 2020 23:12:24 INFO     scan Starting BLE scan
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered 64:8A:9D:18:B5:72 (None)
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered 65:A5:A5:DF:DB:76 (None)
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered 61:9F:04:88:CC:68 (None)
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered 00:09:B0:36:62:60 (None)
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered 36:D5:40:70:BF:DC (None)
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered 43:25:D9:A5:26:F7 (None)
Thu, 01 Oct 2020 23:12:34 INFO     scan Discovered A4:C1:38:3C:43:0E (0131980E433C38C1A4)
Thu, 01 Oct 2020 23:12:34 INFO     scan Found 7 BLE devices
Thu, 01 Oct 2020 23:12:35 INFO     connect Connecting to A4:C1:38:3C:43:0E with timeout=8
Thu, 01 Oct 2020 23:12:35 DEBUG    get_handle Looking up handle for characteristic 00008a22-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a00-0000-1000-8000-00805f9b34fb, value handle: 0x3
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a01-0000-1000-8000-00805f9b34fb, value handle: 0x5
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a04-0000-1000-8000-00805f9b34fb, value handle: 0x7
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a25-0000-1000-8000-00805f9b34fb, value handle: 0xb
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a27-0000-1000-8000-00805f9b34fb, value handle: 0xd
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a26-0000-1000-8000-00805f9b34fb, value handle: 0xf
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a29-0000-1000-8000-00805f9b34fb, value handle: 0x11
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00002a28-0000-1000-8000-00805f9b34fb, value handle: 0x13
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00008a82-0000-1000-8000-00805f9b34fb, value handle: 0x16
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00008a21-0000-1000-8000-00805f9b34fb, value handle: 0x19
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00008a22-0000-1000-8000-00805f9b34fb, value handle: 0x1c
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00008a20-0000-1000-8000-00805f9b34fb, value handle: 0x1f
Thu, 01 Oct 2020 23:12:36 DEBUG    _save_charecteristic_callback Found characteristic 00008a81-0000-1000-8000-00805f9b34fb, value handle: 0x21
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a22-0000-1000-8000-00805f9b34fb handle=28>
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Looking up handle for characteristic 00008a82-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a82-0000-1000-8000-00805f9b34fb handle=22>
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Looking up handle for characteristic 00008a21-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a21-0000-1000-8000-00805f9b34fb handle=25>
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Looking up handle for characteristic 00008a81-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a81-0000-1000-8000-00805f9b34fb handle=33>
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Looking up handle for characteristic 00008a21-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a21-0000-1000-8000-00805f9b34fb handle=25>
Thu, 01 Oct 2020 23:12:37 DEBUG    char_write_handle Sending cmd=char-write-cmd 0x1a 0200
Thu, 01 Oct 2020 23:12:37 INFO     char_write_handle Sent cmd=char-write-cmd 0x1a 0200
Thu, 01 Oct 2020 23:12:37 INFO     subscribe Subscribed to uuid=00008a21-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Looking up handle for characteristic 00008a22-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a22-0000-1000-8000-00805f9b34fb handle=28>
Thu, 01 Oct 2020 23:12:37 DEBUG    char_write_handle Sending cmd=char-write-cmd 0x1d 0200
Thu, 01 Oct 2020 23:12:37 INFO     char_write_handle Sent cmd=char-write-cmd 0x1d 0200
Thu, 01 Oct 2020 23:12:37 INFO     subscribe Subscribed to uuid=00008a22-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Looking up handle for characteristic 00008a82-0000-1000-8000-00805f9b34fb
Thu, 01 Oct 2020 23:12:37 DEBUG    get_handle Found <Characteristic uuid=00008a82-0000-1000-8000-00805f9b34fb handle=22>
Thu, 01 Oct 2020 23:12:37 DEBUG    char_write_handle Sending cmd=char-write-cmd 0x17 0200
Thu, 01 Oct 2020 23:12:37 INFO     char_write_handle Sent cmd=char-write-cmd 0x17 0200
Thu, 01 Oct 2020 23:12:37 INFO     subscribe Subscribed to uuid=00008a82-0000-1000-8000-00805f9b34fb
[... cut ...]
Thu, 01 Oct 2020 23:13:22 INFO     processIndication {'timestamp': 1601590364, 'valid': True, 'weight': 89.5, 'person': 2}
Thu, 01 Oct 2020 23:13:22 INFO     receive_notification Received notification on handle=0x1c, value=0x6f5c1939140264074cf1edf15df120f0000000
Thu, 01 Oct 2020 23:13:22 INFO     processIndication {'timestamp': 1601590364, 'tbw': 49.3, 'person': 2, 'valid': True, 'fat': 33.2, 'muscle': 34.9, 'kcal': 1892, 'bone': 3.2}
Thu, 01 Oct 2020 23:13:50 INFO     <module> Could not disconnect...
Thu, 01 Oct 2020 23:13:50 INFO     <module> Done receiving data from scale
```

[image]

[image]

keptenkurk, 2020-10-02:
Real glad to see your progress! The line
**Thu, 01 Oct 2020 23:12:37 INFO     char_write_handle Sent cmd=char-write-req 0x21 0255193914**
sends the "magic" code to the scale to instruct it to send it's 30 measurement memory (each consisting of 2 data strings).
Since there is no "i'm done" signal, the program waits for 30 seconds listening for any data.
For some obscure reason this command is not always understood and not all data is received. Hence an "unreliable data received" message. But the good thing is: your weight was stored in the 30 measurement buffer associated with your "person#", so on the next go you will get it. This requires the receiving end to discard duplicates received.
The magic code is 02 (hex) followed by the time (in seconds since 01/01/2010 presented in Hex, most significant last) like in the above log: **0255193914** --> hex 14391951 = 339286353 sec = Thursday, 01-Oct-20 22:12:33 UTC. This time is also used to sync the scale's realtime clock. Maybe the command is ignored when the deviation is too big? Who will tell.
Talking about duplicates: The scale will sometimes send weighings twice which are filtered (shown in the log as duplicates).
Cool to see the MQTT plugin in action! (never used it myself)

Caroga, 2020-10-04:
Yeah I figured that would be the case. I am very glad that this is working, I am wondering about other BS444 users if they would have experienced this problem as well or that mine is a updated/outdated version of the hardware? 
Who knows..
If you like I could add an addition to the script and a wiki entry to cover this?
Also it's very cool to use MQTT, however I feel like I would also send to Influx for the sake of history. Looking into this now as I use authenticated connections and the plugin doesn't account for username/password. I'll look into this. 

What plugin do you use?
Also, I'll close this issue as it has been resolved.

closed 2020-10-04 by Caroga