keptenkurk/BS440 #92: Cannot connect to BS444

sgofferj, 2020-10-09:
I have a BS444, IAN 331519, LOT R200218622.
I can find it with hcitool lescan
If I keep it on, I can connect to it with gattool.

However, BS440 times out while trying to connect to it.

Hardware: Pi 3B, internal BT interface.

Relevant log:
´´´
Fri, 09 Oct 2020 13:47:16 INFO     scan Starting BLE scan
Fri, 09 Oct 2020 13:47:26 INFO     scan Discovered A4:C1:38:CD:8A:6A (0131986A8ACD38C1A4)
Fri, 09 Oct 2020 13:47:26 INFO     scan Discovered 65:09:D2:D1:66:34 (None)
Fri, 09 Oct 2020 13:47:26 INFO     scan Discovered name of 65:09:D2:D1:66:34 as Living_room_TV
Fri, 09 Oct 2020 13:47:26 INFO     scan Found 2 BLE devices
Fri, 09 Oct 2020 13:47:26 INFO     connect Connecting to A4:C1:38:CD:8A:6A with timeout=8
Fri, 09 Oct 2020 13:47:34 ERROR    connect Timed out connecting to A4:C1:38:CD:8A:6A after 8 seconds.
Fri, 09 Oct 2020 13:47:34 INFO     connect Connecting to A4:C1:38:CD:8A:6A with timeout=8
Fri, 09 Oct 2020 13:47:42 ERROR    connect Timed out connecting to A4:C1:38:CD:8A:6A after 8 seconds.
Fri, 09 Oct 2020 13:47:42 INFO     connect Connecting to A4:C1:38:CD:8A:6A with timeout=8
´´´

sgofferj, 2020-10-09:
Config:
```
# BS440.ini
# Initialization file for BS440.py
# Change to your preference
# Store this file in the working folder of BS440.py

# Scale properties
# Model: BS410, BS430, BS440, BS444
[Scale] 
ble_address: A4:C1:38:CD:8A:6A
device_name: 0131986A8ACD38C1A4
device_model: BS444

# Program behaviour
# Loglevel: debug | info | critical | error
[Program]
loglevel: debug
logfile: BS440.log
# Enable your plugins here, use the comma separated basename from the plugins/
# Example:
plugins: BS440mqtt
```

sgofferj, 2020-10-09:
Screenshots from bed connect ble scanner:
[image]
[image]
[image]
[image]

sgofferj, 2020-10-09:
Trying some more debugging - learning gatttool at the same time...
```
gatttool -b A4:C1:38:CD:8A:6A -I
[A4:C1:38:CD:8A:6A][LE]> connect
Attempting to connect to A4:C1:38:CD:8A:6A
Connection successful
[A4:C1:38:CD:8A:6A][LE]> primary
attr handle: 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0008, end grp handle: 0x0008 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0009, end grp handle: 0x0013 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x0014, end grp handle: 0x0021 uuid: 000078b2-0000-1000-8000-00805f9b34fb
```

sgofferj, 2020-10-09:
Got it to work. My BS444 doesn't like to be connected with a random address. 
In line 242 of BS444.py I changed
```
addresstype = pygatt.BLEAddressType.random
```
to
```
addresstype = pygatt.BLEAddressType.public
```
With that it works perfectly.

keptenkurk, 2020-10-25:
fixed in commit https://github.com/keptenkurk/BS440/commit/88d574077b6d3b09bf7d596683630d2437de59be

closed 2020-10-25 by keptenkurk