keptenkurk/BS440 #93: Google fit addon can not sent data

letsid, 2020-11-03:
Hi, 
I am trying to sync my data to google fit but i get this erro

Tue, 03 Nov 2020 11:45:00 DEBUG    method URL being requested: PATCH https://fitness.googleapis.com/fitness/v1/users/me/dataSources/raw%3Acom.google.weight%[removed]%3Aunknown%3Aunknown%3A10000001/datasets/[removed]?alt=json
Tue, 03 Nov 2020 11:45:00 ERROR    execute Unable to update Google Fit: Error sending data.
Tue, 03 Nov 2020 11:45:00 INFO     execute Finished plugin: BS440google

as this error is a bit generic I am unsure what is wrong.

rojong, 2020-11-07:
I got the same error,  seems to be caused by code in CheckDataSource BS440google.py?
When the data source does not exist, it should be created by the last line in the function, but the indentation of this line seems to be wrong, when shifing the line to the "left" it works for me!  (you could also add a else: clause)

keptenkurk, 2020-11-08:
There are some upcoming changes to the Google Fit API, but they will not be enforced by april 2021. I will make a separate issue for that as a reminder.
The suggestion of @rojong makes perfectly sense. If the datasource exists (probably in my case) it will never be executed.
I've pushed the fix. Could you try?

letsid, 2020-11-09:
Hi,

Thanks @rojong and @keptenkurk 
Issue is fixed !

closed 2020-11-09 by letsid

keptenkurk, 2020-11-09:
Great to hear that! Happy weighing :-)