keptenkurk/BS440 #95: Scanning just stops working suddenly without any reason.

neforce, 2021-01-13:
Hi there,

There seems to be something odd here. Running the script, after a while (sometimes 30 minutes, sometimes 90 minutes, or anything between), the scan stopped working. It just stops.

E.g. output from the log:

`Tue, 12 Jan 2021 13:59:32 INFO     <module> Configured plugins: BS440influxdb
Tue, 12 Jan 2021 13:59:32 INFO     <module> Loading plugin: BS440influxdb
Tue, 12 Jan 2021 13:59:32 INFO     __init__ Initialising plugin: BS440influxdb
Tue, 12 Jan 2021 13:59:32 INFO     __init__ Read config from: /home/pi/BS440/plugins/BS440influxdb.ini
Tue, 12 Jan 2021 13:59:32 DEBUG    __init__ tags: person
Tue, 12 Jan 2021 13:59:32 INFO     <module> All plugins loaded.
Tue, 12 Jan 2021 13:59:32 INFO     <module> BS440 Started
Tue, 12 Jan 2021 13:59:33 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered ssp br/edr le secure-conn

Tue, 12 Jan 2021 13:59:33 DEBUG    start gatttool_cmd=gatttool -i hci0 -I
Tue, 12 Jan 2021 13:59:33 INFO     scan Starting BLE scan
Tue, 12 Jan 2021 13:59:33 INFO     run Running...
Tue, 12 Jan 2021 13:59:43 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 13:59:43 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 13:59:43 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 13:59:43 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 13:59:43 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 13:59:43 INFO     scan Found 5 BLE devices
Tue, 12 Jan 2021 13:59:43 INFO     scan Starting BLE scan
Tue, 12 Jan 2021 13:59:53 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)`

.... last lines of log ...:

`Tue, 12 Jan 2021 15:30:29 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 15:30:29 INFO     scan Discovered name of XX:XX:XX:XX:XX:XX as Charge 2
Tue, 12 Jan 2021 15:30:29 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 15:30:29 INFO     scan Discovered XX:XX:XX:XX:XX:XX (None)
Tue, 12 Jan 2021 15:30:29 INFO     scan Found 6 BLE devices`

and that's it. Even 

`python BS440.py >bs440.out 2>bs440.err`

does not output anything. Syslogs and other logs either. Tried several things, rebooting, apt-get upgrade everything, nothing.

I'm running Linux raspberrypi 5.4.83-v7+ with Python 2.7.16

Boldfor, 2023-02-24:
@neforce, did you manage to solve the problem? I only read your issue after having filed one myself (https://github.com/keptenkurk/BS440/issues/109).

DjZU, 2023-04-11:
I experienced the same random hanging, did some debugging and managed to find out that the culprit was PyGATT.

The good thing is someone apparently fixed the issue: https://github.com/peplin/pygatt/issues/291

Unfortunately, the commit https://github.com/peplin/pygatt/pull/292/commits/16e15db80360bcb53e0f83b3327dfa387e9249ba has not been pushed yet so the latest release doesn't include it.

The workaround is to apply the commit manually to the file pygatt/backends/gatttool/gatttool.py
I run BS440 in a virtual environment so in my case this file is located at `/opt/BS440/env/lib/python3.10/site-packages/pygatt/backends/gatttool/gatttool.py`
You may run `pip show pygatt` to find the path of the pygatt library on your system.

DjZU, 2023-04-12:
The owner is looking for a new maintainer (see https://github.com/peplin/pygatt/issues/229).
But you can try your luck asking the owner (see https://github.com/peplin/pygatt/issues/195) to push to a new release.

DjZU, 2023-06-22:
It was still hanging so I went the hard way: `touch` a file after every scan and monitor this file with a cron job.

Amend BS440.py
```
	import sys
+++	from pathlib import Path

	found = adapter.filtered_scan(devname, run_as_root=False)
+++	Path('lastscan.ts').touch(mode=0o664, exist_ok=True)
```

Create a small script
`touch /opt/BS440/cron.sh`
`sudo chmod 740 /opt/BS440/cron.sh`
`nano /opt/BS440/cron.sh`
```
#!/bin/bash

# Restart BS440 service if lastscan.ts has not been touched for more than 1 minute 
if [ $(($(date +%s) - $(date +%s -r /opt/BS440/lastscan.ts))) -gt 60 ]; then
	sudo service bs440 restart
fi	

exit 0
```

Set up cron job as root
`sudo crontab -e`
```
* * * * * /opt/BS440/cron.sh
```

The file lastscan.ts should be touched every 10 seconds. The cron job is called every minute and will restart the BS440 service if the lastscan.ts has not been touched for more than a minute.

Now let's rock :metal:

neforce, 2023-09-11:
> It was still hanging so I went the hard way: `touch` a file after every scan and monitor this file with a cron job.
>
> Amend BS440.py
>
> ```
> 	import sys
> +++	from pathlib import Path
>
> 	found = [...]

I picked this up again. Updated the 440 library because some movement happend on github here. Hope this will fix the issue now together with your input.

I'll let you know!

DjZU, 2024-02-08:
I again had to solve 2 other issues that caused the service to stop.

First is probably some system updates which did reset the capabilities set for `btmgmt` and `hcitool`. So I included the commands to set those capabilities in the service file, every time before the script is executed. I updated this comment: https://github.com/keptenkurk/BS440/issues/110#issue-1622383317

Second, I found out that time to time my Bluetooth adapter and HDD ( :scream: ) would fail because of some _Autosuspend USB devices_ feature. Disabling the feature solved the problem.
`sudo nano /etc/default/grub`
```
---	GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"
---	GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.autosuspend=-1"
```
`sudo update-grub`
Reboot
`cat /sys/module/usbcore/parameters/autosuspend` should return -1