keptenkurk/BS440 #96: error when data i received

modular1, 2021-02-24:
Hi,

i keep on getting the following error and bit lost on it. I keep getting the same with and without plugins:

sudo python BS440.py
2021-02-24 18:40:09,413 INFO     <module> No plugins configured.
2021-02-24 18:40:09,418 INFO     <module> BS440 Started
2021-02-24 18:40:09,534 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn 

Traceback (most recent call last):
  File "BS440.py", line 286, in <module>
    indication=True)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/device.py", line 222, in subscribe
    wait_for_response=wait_for_response
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/device.py", line 17, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/device.py", line 48, in char_write_handle
    self._backend.char_write_handle(self, handle, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 50, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 574, in char_write_handle
    self.sendline(cmd)
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 191, in event
    self.wait(event, timeout)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 157, in wait
    raise NotificationTimeout()
pygatt.exceptions.NotificationTimeout: None

Any idea perhaps what it could be or point me in the right direction?

keptenkurk, 2021-02-25:
Is this the complete log? I mean: I woul've expect to see at least a number of scans for BLE devices and the BS440 not to show up before stepping on the scale. The error shows a timeout after the scale has been connected to and the program registers to the notifications which should arrive. Therefore i would expect to see more logging before the exception.
Are you sure you got the right MAC address of the scale (so you're not trying to connect to some other device). Is it a BS440 you're trying? Did you stick with the (meanwhile quite old) version of PyGatt? I know there have been some fixes to that lately, maybe they broke something. BTW i'm thinking of moving to the better supported bleak library if time permits.

modular1, 2021-02-25:
First of all thank you for the reply! I think pygatt was the problem after all....  Now it seems to work. But my quest continues :-) the data does not pop-up in the broker yet, on the the intialised message(suspect i did wrong in the config ;-)  ):

2021-02-25 11:03:42,979 INFO     <module> Loading plugin: BS440mqtt
2021-02-25 11:03:43,232 INFO     <module> All plugins loaded.
2021-02-25 11:03:43,237 INFO     <module> BS440 Started
2021-02-25 11:03:43,354 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn 

2021-02-25 11:04:10,058 INFO     processIndication {'gender': 'male', 'age': 37, 'person': 1, 'valid': True, 'activity': 'normal', 'size': 180}
2021-02-25 11:04:10,073 INFO     <module> Waiting for notifications for another 30 seconds
2021-02-25 11:04:10,077 INFO     processIndication {'timestamp': 1614176741, 'valid': True, 'weight': 81.6, 'person': 1}
2021-02-25 11:04:10,094 INFO     processIndication {'timestamp': 1614176741, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.7, 'muscle': 40.4, 'kcal': 1838, 'bone': 3.4}
2021-02-25 11:04:10,108 INFO     processIndication {'timestamp': 1614181695, 'valid': True, 'weight': 81.8, 'person': 1}
2021-02-25 11:04:10,127 INFO     processIndication {'timestamp': 1614181695, 'tbw': 60.6, 'person': 1, 'valid': True, 'fat': 19.8, 'muscle': 40.3, 'kcal': 1841, 'bone': 3.4}
2021-02-25 11:04:10,147 INFO     processIndication {'timestamp': 1614181799, 'valid': True, 'weight': 81.2, 'person': 1}
2021-02-25 11:04:10,167 INFO     processIndication {'timestamp': 1614181799, 'tbw': 60.8, 'person': 1, 'valid': True, 'fat': 19.5, 'muscle': 40.4, 'kcal': 1832, 'bone': 3.4}
2021-02-25 11:04:10,186 INFO     processIndication {'timestamp': 1614182431, 'valid': True, 'weight': 81.3, 'person': 1}
2021-02-25 11:04:10,206 INFO     processIndication {'timestamp': 1614182431, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.6, 'muscle': 40.4, 'kcal': 1834, 'bone': 3.4}
2021-02-25 11:04:10,226 INFO     processIndication {'timestamp': 1614191767, 'valid': True, 'weight': 82.7, 'person': 1}
2021-02-25 11:04:10,246 INFO     processIndication {'timestamp': 1614191767, 'tbw': 60.3, 'person': 1, 'valid': True, 'fat': 20.1, 'muscle': 40.1, 'kcal': 1853, 'bone': 3.5}
2021-02-25 11:04:10,266 INFO     processIndication {'timestamp': 1614192023, 'valid': True, 'weight': 82.8, 'person': 1}
2021-02-25 11:04:10,286 INFO     processIndication {'timestamp': 1614192023, 'tbw': 60.3, 'person': 1, 'valid': True, 'fat': 20.2, 'muscle': 40.1, 'kcal': 1854, 'bone': 3.5}
2021-02-25 11:04:10,306 INFO     processIndication {'timestamp': 1614193292, 'valid': True, 'weight': 82.7, 'person': 1}
2021-02-25 11:04:10,326 INFO     processIndication {'timestamp': 1614193292, 'tbw': 60.3, 'person': 1, 'valid': True, 'fat': 20.2, 'muscle': 40.1, 'kcal': 1853, 'bone': 3.5}
2021-02-25 11:04:10,346 INFO     processIndication {'timestamp': 1614193396, 'valid': True, 'weight': 82.1, 'person': 1}
2021-02-25 11:04:10,366 INFO     processIndication {'timestamp': 1614193396, 'tbw': 60.6, 'person': 1, 'valid': True, 'fat': 19.8, 'muscle': 40.3, 'kcal': 1845, 'bone': 3.4}
2021-02-25 11:04:10,386 INFO     processIndication {'timestamp': 1614193867, 'valid': True, 'weight': 81.9, 'person': 1}
2021-02-25 11:04:10,406 INFO     processIndication {'timestamp': 1614193867, 'tbw': 60.6, 'person': 1, 'valid': True, 'fat': 19.8, 'muscle': 40.3, 'kcal': 1842, 'bone': 3.4}
2021-02-25 11:04:10,426 INFO     processIndication {'timestamp': 1614194153, 'valid': True, 'weight': 81.6, 'person': 1}
2021-02-25 11:04:10,446 INFO     processIndication {'timestamp': 1614194153, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.7, 'muscle': 40.4, 'kcal': 1838, 'bone': 3.4}
2021-02-25 11:04:10,466 INFO     processIndication {'timestamp': 1614194307, 'valid': True, 'weight': 81.6, 'person': 1}
2021-02-25 11:04:10,486 INFO     processIndication {'timestamp': 1614194307, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.7, 'muscle': 40.4, 'kcal': 1838, 'bone': 3.4}
2021-02-25 11:04:10,506 INFO     processIndication {'timestamp': 1614194844, 'valid': True, 'weight': 81.4, 'person': 1}
2021-02-25 11:04:10,526 INFO     processIndication {'timestamp': 1614194844, 'tbw': 60.8, 'person': 1, 'valid': True, 'fat': 19.6, 'muscle': 40.4, 'kcal': 1835, 'bone': 3.4}
2021-02-25 11:04:10,546 INFO     processIndication {'timestamp': 1614195066, 'valid': True, 'weight': 81.7, 'person': 1}
2021-02-25 11:04:10,566 INFO     processIndication {'timestamp': 1614195066, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.7, 'muscle': 40.4, 'kcal': 1839, 'bone': 3.4}
2021-02-25 11:04:10,586 INFO     processIndication {'timestamp': 1614195412, 'valid': True, 'weight': 81.6, 'person': 1}
2021-02-25 11:04:10,606 INFO     processIndication {'timestamp': 1614195412, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.7, 'muscle': 40.4, 'kcal': 1838, 'bone': 3.4}
2021-02-25 11:04:10,626 INFO     processIndication {'timestamp': 1614196026, 'valid': True, 'weight': 81.3, 'person': 1}
2021-02-25 11:04:10,646 INFO     processIndication {'timestamp': 1614196026, 'tbw': 60.8, 'person': 1, 'valid': True, 'fat': 19.6, 'muscle': 40.4, 'kcal': 1834, 'bone': 3.4}
2021-02-25 11:04:10,666 INFO     processIndication {'timestamp': 1614196116, 'valid': True, 'weight': 81.3, 'person': 1}
[... cut ...]
2021-02-25 11:04:11,226 INFO     processIndication {'timestamp': 1614248863, 'tbw': 60.5, 'person': 1, 'valid': True, 'fat': 19.9, 'muscle': 40.2, 'kcal': 1843, 'bone': 3.4}
2021-02-25 11:04:11,246 INFO     processIndication {'timestamp': 1614251035, 'valid': True, 'weight': 81.4, 'person': 1}
2021-02-25 11:04:11,266 INFO     processIndication {'timestamp': 1614251035, 'tbw': 60.7, 'person': 1, 'valid': True, 'fat': 19.7, 'muscle': 40.4, 'kcal': 1835, 'bone': 3.4}
2021-02-25 11:04:40,094 INFO     <module> Could not disconnect...
2021-02-25 11:04:40,098 INFO     <module> Done receiving data from scale

keptenkurk, 2021-02-25:
I never used the MQTT plugin myself. But you should at least see the plugin starting when proper data is received as per line 316 "if persondata and weightdata and bodydata:" (BS440.py) Maybe throw in some debugging loggings...

Boldfor, 2023-01-12:
@modular1, in case this is still relevant: i got MQTT working like a charm. Let me know if I can help.