keptenkurk/BS440 #98: get timeouts when trying to connect

compujunk, 2021-04-23:
I have a BS445 which according to the medisana support page is the same as the BS440 but that probably means its operation.

When I run BS440.py and put on the scale I get the following in my log file:

`Fri, 23 Apr 2021 12:06:52 INFO     <module> Configured plugins: BS440csv
Fri, 23 Apr 2021 12:06:52 INFO     <module> Loading plugin: BS440csv
Fri, 23 Apr 2021 12:06:52 INFO     <module> All plugins loaded.
Fri, 23 Apr 2021 12:06:52 INFO     <module> BS440 Started
Fri, 23 Apr 2021 12:06:52 INFO     init_ble_mode hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn

Fri, 23 Apr 2021 12:06:53 DEBUG    start gatttool_cmd=gatttool -i hci0 -I
Fri, 23 Apr 2021 12:06:53 INFO     run Running...
Fri, 23 Apr 2021 12:06:53 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 76:9E:2A:3D:47:43 (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 4F:BA:C0:2D:06:2A (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 6D:FF:D2:4C:15:3D (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered name of 6D:FF:D2:4C:15:3D as JBL TUNE125TWS-LE
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 62:B7:8D:6D:69:9A (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 4F:C7:2D:F2:63:E2 (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 55:D2:A9:87:09:7A (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Discovered 3C:5D:4C:1D:C6:BB (None)
Fri, 23 Apr 2021 12:07:03 INFO     scan Found 9 BLE devices
Fri, 23 Apr 2021 12:07:03 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 4F:BA:C0:2D:06:2A (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 76:9E:2A:3D:47:43 (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 6D:FF:D2:4C:15:3D (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 4F:C7:2D:F2:63:E2 (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Discovered 62:B7:8D:6D:69:9A (None)
Fri, 23 Apr 2021 12:07:13 INFO     scan Found 7 BLE devices
Fri, 23 Apr 2021 12:07:13 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:07:23 INFO     scan Discovered 4F:BA:C0:2D:06:2A (None)
Fri, 23 Apr 2021 12:07:23 INFO     scan Discovered 76:9E:2A:3D:47:43 (None)
Fri, 23 Apr 2021 12:07:23 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:07:23 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:07:23 INFO     scan Discovered 3C:5D:4C:1D:C6:BB (None)
Fri, 23 Apr 2021 12:07:23 INFO     scan Discovered 62:B7:8D:6D:69:9A (None)
Fri, 23 Apr 2021 12:07:23 INFO     scan Found 6 BLE devices
Fri, 23 Apr 2021 12:07:23 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:07:33 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:07:33 INFO     scan Discovered 4F:BA:C0:2D:06:2A (None)
Fri, 23 Apr 2021 12:07:33 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:07:33 INFO     scan Discovered 76:9E:2A:3D:47:43 (None)
Fri, 23 Apr 2021 12:07:33 INFO     scan Found 4 BLE devices
Fri, 23 Apr 2021 12:07:33 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:07:43 INFO     scan Discovered 76:9E:2A:3D:47:43 (None)
Fri, 23 Apr 2021 12:07:43 INFO     scan Discovered 4F:BA:C0:2D:06:2A (None)
Fri, 23 Apr 2021 12:07:43 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:07:43 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:07:43 INFO     scan Discovered 77:F6:C3:CA:0D:B1 (None)
Fri, 23 Apr 2021 12:07:43 INFO     scan Discovered 52:2A:C2:5F:68:86 (None)
Fri, 23 Apr 2021 12:07:43 INFO     scan Found 6 BLE devices
Fri, 23 Apr 2021 12:07:43 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 4F:BA:C0:2D:06:2A (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 76:9E:2A:3D:47:43 (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 77:56:E6:33:9D:9D (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 66:68:2F:55:C8:33 (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Discovered 4B:7A:22:9E:7B:0B (None)
Fri, 23 Apr 2021 12:07:54 INFO     scan Found 7 BLE devices
Fri, 23 Apr 2021 12:07:54 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:08:04 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:08:04 INFO     scan Discovered 4B:7A:22:9E:7B:0B (None)
Fri, 23 Apr 2021 12:08:04 INFO     scan Discovered 47:84:C5:59:1A:4A (None)
Fri, 23 Apr 2021 12:08:04 INFO     scan Discovered 66:68:2F:55:C8:33 (None)
Fri, 23 Apr 2021 12:08:04 INFO     scan Discovered DC:66:1F:AE:C1:F5 (013197F5C1AE1F66DC)
Fri, 23 Apr 2021 12:08:04 INFO     scan Discovered 62:BE:F0:5D:2B:FF (None)
Fri, 23 Apr 2021 12:08:04 INFO     scan Found 6 BLE devices
Fri, 23 Apr 2021 12:08:04 INFO     connect Connecting to DC:66:1F:AE:C1:F5 with timeout=8
Fri, 23 Apr 2021 12:08:12 ERROR    connect Timed out connecting to DC:66:1F:AE:C1:F5 after 8 seconds.
Fri, 23 Apr 2021 12:08:12 INFO     connect Connecting to DC:66:1F:AE:C1:F5 with timeout=8
Fri, 23 Apr 2021 12:08:20 ERROR    connect Timed out connecting to DC:66:1F:AE:C1:F5 after 8 seconds.
Fri, 23 Apr 2021 12:08:20 INFO     connect Connecting to DC:66:1F:AE:C1:F5 with timeout=8
Fri, 23 Apr 2021 12:08:28 ERROR    connect Timed out connecting to DC:66:1F:AE:C1:F5 after 8 seconds.
Fri, 23 Apr 2021 12:08:28 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:08:28 ERROR    scan BLE adapter requires reset after a scan as root- call adapter.reset()
Fri, 23 Apr 2021 12:08:34 INFO     scan Starting BLE scan
Fri, 23 Apr 2021 12:08:44 INFO     scan Discovered 62:BE:F0:5D:2B:FF (None)
Fri, 23 Apr 2021 12:08:44 INFO     scan Discovered 4B:7A:22:9E:7B:0B (None)
Fri, 23 Apr 2021 12:08:44 INFO     scan Discovered 75:1D:49:25:D4:5B (None)
Fri, 23 Apr 2021 12:08:44 INFO     scan Discovered 73:87:FC:7D:B6:DE (None)
Fri, 23 Apr 2021 12:08:44 INFO     scan Discovered 4F:C7:2D:F2:63:E2 (None)
Fri, 23 Apr 2021 12:08:44 INFO     scan Discovered 5F:08:E3:13:33:F8 (None)
Fri, 23 Apr 2021 12:08:44 INFO     scan Found 6 BLE devices
`

keptenkurk, 2021-04-24:
Could you give it a try specifiying BS444 as the scale model?  The BS444 uses public adressing (where random adressing would cause your "unable to connect" issue. Also the later models start the clock at 1/1/2010 (instead of standard unix timestamp).

compujunk, 2021-04-24:
still timeouts with BS444 as scale model
```
Sat, 24 Apr 2021 21:21:35 INFO     scan Starting BLE scan
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 7A:72:F3:FC:72:42 (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 4D:93:9C:61:8E:A6 (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 51:B9:4D:95:77:75 (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 73:9C:4B:DA:5A:C5 (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 61:4B:40:59:48:BB (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered DC:66:1F:AE:C1:F5 (013197F5C1AE1F66DC)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 71:52:2F:BD:46:17 (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Discovered 74:38:DD:5C:4F:B9 (None)
Sat, 24 Apr 2021 21:21:45 INFO     scan Found 8 BLE devices
Sat, 24 Apr 2021 21:21:45 INFO     connect Connecting to DC:66:1F:AE:C1:F5 with timeout=8
Sat, 24 Apr 2021 21:21:53 ERROR    connect Timed out connecting to DC:66:1F:AE:C1:F5 after 8 seconds.
Sat, 24 Apr 2021 21:21:53 INFO     connect Connecting to DC:66:1F:AE:C1:F5 with timeout=8
Sat, 24 Apr 2021 21:22:01 ERROR    connect Timed out connecting to DC:66:1F:AE:C1:F5 after 8 seconds.
Sat, 24 Apr 2021 21:22:01 INFO     connect Connecting to DC:66:1F:AE:C1:F5 with timeout=8
Sat, 24 Apr 2021 21:22:09 ERROR    connect Timed out connecting to DC:66:1F:AE:C1:F5 after 8 seconds.
Sat, 24 Apr 2021 21:22:09 INFO     scan Starting BLE scan
Sat, 24 Apr 2021 21:22:10 ERROR    scan BLE adapter requires reset after a scan as root- call adapter.reset()
Sat, 24 Apr 2021 21:22:15 INFO     scan Starting BLE scan
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered DC:66:1F:AE:C1:F5 (013197F5C1AE1F66DC)
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered 73:9C:4B:DA:5A:C5 (None)
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered 7A:72:F3:FC:72:42 (None)
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered 61:4B:40:59:48:BB (None)
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered 51:B9:4D:95:77:75 (None)
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered 4D:93:9C:61:8E:A6 (None)
Sat, 24 Apr 2021 21:22:25 INFO     scan Discovered 71:E1:C9:20:5E:B2 (None)
Sat, 24 Apr 2021 21:22:25 INFO     scan Found 7 BLE devices
```

keptenkurk, 2021-04-24:
Got the Pi close enough to the scale? Discovery works at large distance, connecting needs it to be at max 1..2 m
Are you able to connect to it using gattool from the commandline manually? (is not easy as the scale is only connectable right after processing the measurement).  The code uses these gatttool commands in background.

compujunk, 2021-04-24:
at first I did but now suddenly I don't see anymore.
Tomorrow I will reinstall everything and see if it works again if that works I will try BS444 again.
I'll let you know if it works then or if I'm still having issues.

compujunk, 2021-04-25:
i have reinstalled everything but it still doesn't work. now i read somewhere else that the bluetooth of the Pi3 + is not very stable so i'm going to try it first with a BT dongle but i have to purchase it first. I will come back to this later