keptenkurk/BS440 #99: Python 3

sgofferj, 2021-05-17:
Many if not most Linux distros don't support Python 2 anymore. It would be nice if the code could be updated to Python 3, so it runs on a newer distro.

Tankyspanky, 2021-08-19:
Yeah maybe time for a "update" :) 
--------------------------------------
 BS440.py
/home/aad/BS440/BS440.py:192: DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in future versions. Use ConfigParser directly instead.
  config = SafeConfigParser()

And i also get a error on the BS440Domoticz.py on line 36 it looks like there is something missing there ??

I hope you read this and are willing to answer ;) 

Because i would love to see it work

Tankyspanky, 2021-09-06:
Jammer zag eruit als een mooi project om meer te kunnen integreren, maar het blijft achter in de tijd ...
Python2 vs Python3

sgofferj, 2021-09-07:
@Tankyspanky It's obvious that either the project is abandoned or the author is otherwise busy. Spamming the issue doesn't help in either case.

keptenkurk, 2021-09-19:
Hi guys. 
It has been over 6 years since i prgrammed this BS440. Being an absolute NOOB on BLE, the BS440 and even Python programming alltogether i took me much time, but every minute was well spent since i learned so much googeling.
The script is still in daily use at my home, but keeping it up to date is not at the top of my list currently.
I even wonder if PyGatt is still usable and the right tool to use nowadays. And surely Python3 should be the interpreter to run on,

But i'm sure that with little effort, the syntax error in the comment above can be fixed.
Pull requrests are welcomed.

aeff81, 2022-01-31:
Here's my patch to get the script running on Python 3.9:
diff --git a/BS440.py b/BS440.py
index 8049e2a..9698216 100644
--- a/BS440.py
+++ b/BS440.py
@@ -1,7 +1,7 @@
 from __future__ import print_function
 import pygatt.backends
 import logging
-from ConfigParser import SafeConfigParser
+from configparser import SafeConfigParser
 import time
 import subprocess
 from struct import *
@@ -55,12 +55,12 @@ def decodePerson(handle, values):
 
 def sanitize_timestamp(timestamp):
     retTS = 0
-    if timestamp + time_offset < sys.maxint:
+    if timestamp + time_offset < sys.maxsize:
         retTS = timestamp + time_offset
     else:
         retTS = timestamp
 
-    if timestamp >= sys.maxint:
+    if timestamp >= sys.maxsize:
         retTS = 0
 
     return retTS

creatvty, 2022-07-30:
Hey, thanks for the help. I was able to make it work. Also I added now Auto Discovery for Homeassistant to the mqtt plugin https://github.com/creatvty/BS440-HA-AD

Boldfor, 2023-01-07:
> Hey, thanks for the help. I was able to make it work.

@creatvty, can you confirm that your new version is running under Python 3.9? I keep getting the following error:
```
Traceback (most recent call last):
  File "/home/pi/BS440-HA-AD/BS440.py", line 204, in <module>
    config.get('Program', 'loglevel').upper(),
  File "/usr/lib/python3.9/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.9/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'Program'
```

Boldfor, 2023-01-12:
@creatvty, please see my PR (#2) on fixing an issue around SafeConfigParser.

creatvty, 2023-01-12:
Hey @Boldfor so you made it work? Cool stuff. I'll take a look.

Boldfor, 2023-01-22:
@sgofferj, should be fixed and merged. Please have a look and close the issue.

sgofferj, 2023-01-22:
I'll dig out an RPi and test it. Might take a moment, though because I repurposed the one I used for reading the scale.

sgofferj, 2024-08-30:
Apologies for the delay. I finally dug out an RPi3 and installed it and it works. Thank you!

closed 2024-08-30 by sgofferj