keptenkurk/BS440 #102: Division by zero in BS440csv.py

aeff81, 2022-01-31:
There's a potential "division by 0" in BS440csv.py if size = 0 (Guest-User for example).
Due to my lack of Python skills, I solved it the ugly way. But I would suggest that somebody, who knows what he's doing, adds a check 
if (size == 0) {
  size = 1
}
Here's my ugly patch, just to show the location of the Problem:
```
diff --git a/plugins/BS440csv.py b/plugins/BS440csv.py
index 0131124..a2a62e6 100644
--- a/plugins/BS440csv.py
+++ b/plugins/BS440csv.py
@@ -32,6 +32,7 @@ class Plugin:
                calculateddata = []
                datetimedata = []
                size = persondata[0]['size'] / 100.00
+               size = 1
                for i, e in list(enumerate(weightdata)):
                        bmiDict = {}
                        bmiDict['bmi'] = round(weightdata[i]['weight'] / (size * size), 1)

```

Boldfor, 2023-01-07:
This is not only relevant for „guest users“. I have the BS 444, and sometimes the „user detection“ fails (or does not even start), which results in „only weight being sent“ (with everything being blank). Also in these (multiple) cases they script stops with „division by 0“.

Boldfor, 2023-01-12:
@aeff81, not sure whether I know what I'm doing ;) but I added a PR for this bug (https://github.com/creatvty/BS440-HA-AD/pull/2) which fixed it for me.

@creatvty, please have a look.

Boldfor, 2023-01-22:
@aeff81, is merged and should be fixed. Maybe you can check and close the issue.

closed 2025-02-17 by aeff81