keptenkurk/BS440 #110: blueman-applet crash solved

DjZU, 2023-03-13:
I am running BS440 for years as a service on a machine running 24/7 and operating Ubuntu/XFCE.
Across distribution upgrades, and maybe when I started to run this as a service instead of invoking the script directly in the command line, I went through instabilities and an annoying crash pop-up (https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1860851).

After reading pygatt code (https://github.com/peplin/pygatt/blob/master/pygatt/backends/gatttool/gatttool.py), I managed to solve the issue by running the script as a normal user without privileges instead of running the script as root. This way, resetting the adaptor will not be required anymore, and in case it does, the service will take care of that.

I believe this may also fix other issues like #109 and #95.

Here are the steps:

Enable normal users to perform LE scanning
``sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hcitool` ``

Amend BS440.py code
```
		# wait for scale to wake up and connect to it
---		found = adapter.filtered_scan(devname)
+++		found = adapter.filtered_scan(devname, run_as_root=False)

---		# reset adapter when (see issue #33)
---		adapter.reset()
+++		#adapter.reset()
+++		log.info('Adaptor error while scanning for devices. Will exit. If run via systemd, adapter will be reset before automatic restart of this script.')
+++		sys.exit(2)

log.info('BS440 Started')
--- if not init_ble_mode():
--- sys.exit()
+++ #if not init_ble_mode():
+++ #	sys.exit(1)

adapter = pygatt.backends.GATTToolBackend()
--- adapter.start()
+++ adapter.start(reset_on_start=False)
```

Amend ../BS440/dist/init/linux-systemd/bs440.service
```
[Service]
+++ User=<username>
+++ Group=<username>
+++ # Restart if process terminates (will terminate with error if bluetooth adapter needs reset)
+++ Restart=on-failure
+++ # Reset bluetooth adapter
+++ # Prefix '+' will run command with full privileges regardless of privilege restrictions configured with 'User=' and 'Group='
+++ #ExecStopPost=+systemctl restart bluetooth
+++ ExecStopPost=+hciconfig bluetooth reset
+++ # Turn Bluetooth Low Energy on
+++ ExecStartPre=+btmgmt le on
```

closed 2023-03-13 by DjZU

Boldfor, 2023-03-20:
Thanks a lot!

Am currently checking whether I can reproduce [the problem I previously experienced](https://community.openhab.org/t/use-case-connecting-a-bathroom-body-fat-scale-e-g-medisana-bs440-via-bluetooth-mqtt-to-openhab-step-by-step-instruction/143296/3?u=cplant). If yes, I'll try your fix and eventually create a PR for that. While I kept having problems, another user (with the same code) [did not experience any problem](https://community.openhab.org/t/use-case-connecting-a-bathroom-body-fat-scale-e-g-medisana-bs440-via-bluetooth-mqtt-to-openhab-step-by-step-instruction/143296/12?u=cplant).

Boldfor, 2023-04-02:
Am still waiting for BS440 to crash according to the standard installation guideline, to try your fix, ..., but it doesn't crash anymore. Weird...