keptenkurk/BS440 #114: Add multiple people entities

dimaguy, 2023-08-15:
Currently only one person using the scale is supported at a time, but a scale like the BS444 can support up to 8 people and determine automatically who is getting measured. This could translate into multiple entities in MQTT/Home Assistant (and other plugins)

DjZU, 2023-08-15:
BS440csv and BS440flask (graphic displays from the CSV data served on a local webserver) do support several persons. I don't know for the other plugins though.

Boldfor, 2023-08-15:
I have the MQTT-plug-in also working with multiple people.

dimaguy, 2023-08-15:
Ooh I see, I just didn't know because I saw nothing about it in the readme (maybe a point to improve?) I'll be closing this issue now

closed 2023-08-15 by dimaguy

Boldfor, 2023-08-15:
> maybe a point to improve?

Good idea. Added a couple of things to the readme and inline documentation as well already.